LibreCAD/LibreCAD #1969: Edit Layer dialog in Layer Tree shows two color sets for Layer Types

Questions2000, 2024-11-29:
BUG REPORT

### Observed Behavior
“Layer Tree” widgets “Edit Layer” dialog shows double color sets when activate certain “Layer Type” radio buttons

### Steps To Reproduce
-Open "Layer Tree" widget
-Create a new layer
-Open the “Edit Layer” dialog for that new layer
-Click on one of the following three items in “Layer Type” section
Informational
Dimensions
Alternative Position
-Click on the Color drop down
-You will notice that this list is showing double list of layer colors, one after the other.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-207-g7838da61.AppImage

sand1024, 2024-11-30:
@Questions2000  wow! that's a really good point, thank you for reporting it. 

This is bug which was easy to make (a kind of mistyping in the code) yet not so easy to note.

sand1024, 2024-12-18:
@Questions2000 
addressed in last AppImage

Questions2000, 2024-12-20:
Tested and appears to be corrected.
Thank You for fixing this reported issue.

closed 2024-12-20 by Questions2000