LibreCAD/LibreCAD #1973: Undo and Redo Relative Zero Point

ttk, 2024-12-10:
I expect that the relative zero point to respect the undo and redo system.  Currently when you undo, the relative zero point stays where it is even though it isn't locked. 

To reproduce, run the commands:
```
rectangle; 
1,1; 2,2;
4,4; 5,5;
undo
```

In the above case, the relative zero point will at `5,5` instead of the expected point `2,2` which was the position of the relative zero point prior to the creation of the second rectangle.

Not reverting the relative zero point forces me to manually set it again, which breaks my workflow, and slows me down.

Tested with `2.2.0` and `2.2.1_rc-latest`.

dxli, 2024-12-10:
Current design, relative zero is not an entity or geometry object, and the undo redo only works on geometry objects.

The explains the current behavior. A new design is needed include undo support for the relative zero object.

sand1024, 2024-12-11:
@ttk 
Yes, you're correct, that's annoying
@dxli 
yes, currently moving relative point is not included into undo cycle. 
For some commands (that includes own undo, like polyline - that's not necessary and may be solved on the action level, yet for most of actions - indeed, it requires extension of undo design. 
Actually, I considered changing this already, so hope I'll include this into one of next update.

sand1024, 2025-01-01:
@ttk
Now it should be undoable - this feature is included into the last AppImage, please check and close the ticket if it's fine now.

ttk, 2025-01-03:
@sand1024 This is fantastic!  However, I couldn't find a working Windows build to test it.  I'll check back again in a few days.

closed 2026-09-26 by sand1024