LibreCAD/LibreCAD #1976: Command Line - Capitalize Keycode text in log pane

Questions2000, 2024-12-15:
### Expected Behavior
Make Keycodes text start with capital letter to match how other items show in the CL log pane.
For example, commands entered in the CL field show up as
`Command:`
`Unknown command:`

When you type in a Keycode, for example `li`, would expect to show up as
`Keycode: li (line2p)`

If you type in an invalid Keycode, for example `ko` it shows up as
`Invalid keycode: ko`

### Observed Behavior
When you type in a Keycode, for example `li` it shows up as
`keycode: li (line2p)`

If you type in an invalid Keycode, for example `ko` it shows up as
`invalid keycode: ko` 

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-207-g7838da61.AppImage

Gary-Wheeler, 2024-12-30:
I just finished a one page diagram of Command Line keycode, and used UPPERCASE for all the two letter commands! As an architect I am used to typing in all caps as a standard! Oops.  Am new at this, the list was intended I thought for 2.2.0 
Have not yet looked at 2.2.1 or 2.2.2

[LibreCADCommands08.pdf](https://github.com/user-attachments/files/18272845/LibreCADCommands08.pdf)

Questions2000, 2025-03-21:
I did not realize anyone responded to this, I am not sure that this is relevant to this topic though?
If anything maybe more relevant to my post 1975?

Gary-Wheeler, 2025-03-23:
To continue the thread on capitalization of Commands, here is a one page pdf of what I could find of the LibreCAD Commands in LOWER CASE just in case capitalization might mess things up. This is version 08 from January. 

Just for giggles, I am also including a LibreOffice spreadsheet that is the basis of this. Similar name. In case anyone needs to play with it. I'd like to have the page added to the User Manual at some time at at present its a little over my head. 

[LibreCADCommands08.ods](https://github.com/user-attachments/files/19411903/LibreCADCommands08.ods)

[LibreCADCommands08.pdf](https://github.com/user-attachments/files/19411884/LibreCADCommands08.pdf)

closed 2025-04-23 by Questions2000