LibreCAD/LibreCAD #1979: Command Line - Clear field if invalid keycode is entered

Questions2000, 2024-12-15:
### Expected Behavior
When using Keycodes, if an invalid Keycode is entered, have the command line field be cleared out automatically.

### Observed Behavior
Currently if an invalid Keycode is entered, the two letters remain in the field and user has to delete them manually.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-207-g7838da61.AppImage

If anyone is interested a list of all my Command Line based posts can be found here https://github.com/LibreCAD/LibreCAD/issues/1974

dxli, 2024-12-18:
does it mean editing is not possible after a typo?

Questions2000, 2024-12-19:
The field can still be edited after the typo/invalid command.
Just feels like the field should be cleared out after allowing the user to just type in the keycode again.
Even if user got the first letter right, they would still have to hit Backspace key and another letter key which would be 2 key presses.  If typed two wrong letters, then equals 3 key presses.
Also Backspace makes user have to take hands out of typing position on keyboard.