LibreCAD/LibreCAD #1980: Add a hotkey to toggle Keycodes on and off

Questions2000, 2024-12-15:
Would anyone consider adding a hotkey to toggle Keycodes on and off in the “Keyboard Shortcuts” window.

Thank You

sand1024, 2024-12-16:
@Questions2000 

mmm... don't get your point, sorry. Could you please expand on it? Which keycodes to toggle? Where? etc.

Thank you

Questions2000, 2024-12-20:
@sand1024 
Sure, I am not referring to specific keycodes, I am referring to the actual setting that allows user to turn on/off the use of "Keycodes"
Found in the icon/button to the right of the CL field, in context menu that it opens up.
Allowing this to be assigned to a hotkey would allow user to quickly be able to turn this feature on/off rather than having to mouse over to the button and navigate in a context menu to do so.
Keycodes are convenient but user may need/want to type certain commands at times.
Thank You

sand1024, 2024-12-20:
@Questions2000 
Oh, got it, thank you.