LibreCAD/LibreCAD #1981: Capitalize Order icon flyout menu items in Categories toolbar

Questions2000, 2024-12-15:
POSSIBLE GUI UPDATE

### Expected Behavior
To have “Order” icon flyout menu items in Categories toolbar to be capitalized to match other flyout menu items

Move to top
Move to bottom
Raise over entity
Lower after entity

OR

Move to Top
Move to Bottom
Raise over Entity
Lower after Entity

### Observed Behavior
Currently shows
move to top
move to bottom
raise over entity
lower after entity

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-207-g7838da61.AppImage

sand1024, 2024-12-22:
will be fixed in next image

Questions2000, 2024-12-31:
Tested and looks good
Thank You

closed 2024-12-31 by Questions2000