LibreCAD/LibreCAD #1984: If hit Escape key to close Properties dialog the cursor remains a crosshair

Questions2000, 2024-12-20:
### Expected Behavior
To return mouse cursor to Arrow after the Properties dialog is closed, when use Escape key to close.
Like how the "Cancel" and "OK" buttons do.

### Observed Behavior
When you double click on an existing drawing object to open its Properties dialog, LC enters a "Modify" mode that turns the Arrow mouse cursor into Crosshair cursor, right before the Properties dialog opens.
If you hit the "Cancel" or "OK" button the mouse cursor returns back to an Arrow.
However, if you hit your "Escape" key the mouse cursor remains a Crosshair, have to hit Escape key again.

### Steps To Reproduce
-Not really, but might help to turn ON the following setting in Application Preferences
"Hide cursor when snapping" (You may have to restart LC after this)
-Draw an item (Line, circle, etc.)
-Double click on that item to open its Properties dialog
-Notice right before the Properties dialog opens your mouse cursor turns from Arrow to Crosshair
-If you click on "Cancel" or "OK" buttons the mouse cursor goes back to an Arrow
-If you hit your "Escape" key to close the dialog, then your mouse cursor remains in Modify Crosshair mode
-Have to hit "Escape" again to get Arrow cursor back

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-217-gff06defb.AppImage

sand1024, 2024-12-22:
@Questions2000 
> -If you hit your "Escape" key to close the dialog, then your mouse cursor remains in Modify Crosshair mode

Hm... that's weird. I'm unable to reproduce this under Windows, here the cursor is restored properly. 
Yet I'll will check under Linux too.

However, potentially it might be that such behavior is related to some specific window manager.

Questions2000, 2025-01-02:
Appears to work perfectly
Both issues were corrected with this fix
1983 & 1984
Thank You

closed 2025-01-02 by Questions2000