LibreCAD/LibreCAD #1985: Command Line - Remove header when inserted in top or bottom dock

Questions2000, 2024-12-20:
GUI UPDATE

### Observed Behavior
The Command Line widgets header shows up on the left side of the widget by default.
There is already a lot of activity/information happening all in that same general area of the widget when in top or bottom dock.

### Expected Behavior
Suggesting that when the CL is inserted in the top or bottom dock to remove the header completely from the widget.  It does not really serve a purpose and could give the CL a cleaner appearance.
Usually once docked, the user does not typically move the CL around.
If user needs to, they can go into the icon context menu to the right of the CL field and choose “Float” which then would allow them to move the CL to a different dock/location.

### Links
This post is a similar update to the one mentioned in the following post
LINK

If anyone is interested a list of all my Command Line based posts can be found here https://github.com/LibreCAD/LibreCAD/issues/1974

sand1024, 2025-03-29:
@Questions2000 
Hm... well, that's interesting suggestion - yet such approach will make cmd Widget unique and behave differently than other widgets. 
Plus, I'm not sure it will be obvious to the user how to make the widget be floated/redocked... 

Yet anyway, I'll think about this.  And could you share a screenshot of your typical LC screen layout? It might be that it will be inspiring. 

> Usually once docked, the user does not typically move the CL around.

That's exactly that was annoyed me. So now it's not true - there could be different workspaces for different tasks. 

https://youtu.be/FwvXs1eDYBA

Questions2000, 2025-04-22:
I apologize I did not realize anyone responded to this post.
I just tested out LibreCAD-2.2.2_alpha1-398-g65f3c5fa7.AppImage
I know I do not get a say in the matter, but this bums me out a little.
As mentioned, when docked in the bottom dock, a full header at the top of the CL takes up space/height of screen but does not really serve any purpose for the space.
The "Command line" widget header (which I would capitalize the "L" in "line" to match the other proper case used in other widgets, I can mention in new post if needed) above white log window, now when used in the bottom dock is close / mimics how the "Enter Command:" text looks right above the white command line field does.
So it kind of looks like two fields that can be typed in.
To me CL on bottom is standard, I could not use in side dock, I look down to type in field, would feel strange to look to right side dock where other widgets are to type in commands, but I might be the only one.

I have other things to mention but will wait to see the response I get on this, if they are still relevant.

Thank You

Questions2000, 2025-04-23:
@sand1024
Thank you for your reply.
Just to clarify, I think the decision makes sense to make all headers the same orientation, as requested in 1986.
I apologize, when I originally made this specific post focusing on having CL be the only widget in the bottom dock (what I call "Standard" CAD layout) which therefore could have its header removed, but forgot that LC allows for multiple widgets to be placed side by side in the bottom dock. 
So I agree, this would not be correct to have as the default for CL in bottom dock, would not look/function well.
However, the CL header at top when used in this "standard" (common) layout could create some graphical issues.  Might not be worth adding but I have an idea for this I will make a separate post for if interested.
Also I will be making a (closed) post about how to generate the "standard" (common) layout I am referring to per your question.
Thanks again

closed 2025-04-23 by Questions2000