LibreCAD/LibreCAD #1986: Command Line - Move header to top of widget when inserted into side dock

Questions2000, 2024-12-20:
GUI UPDATE

### Observed Behavior
The Command Line widgets header shows up on the left side of the widget which when docked in side dock can look a little strange because not consistent with the other widgets.

### Expected Behavior
Suggesting that when CL is inserted into side dock to have its header appear at the top of the widget like how all the other widgets currently appear.
Can then give the header a full name “Command Line”

### Links
This post is a similar update to the one mentioned in the following post #1985

If anyone is interested a list of all my Command Line based posts can be found here https://github.com/LibreCAD/LibreCAD/issues/1974

Questions2000, 2025-03-10:
I personally use CL under the drawing pane so I am more interested in my other request post 1985, but I came across something interesting.  I was doing an image search for LC for something else and I noticed that in a few images there was something different.
I followed one of the images to this link which is from the LC forum
https://forum.librecad.org/Ortho-mode-in-command-line-td5716412.html
if you scroll down to the bottom there is a picture that shows a red arrow pointing at ortho button.
If you look at the CL widget docked in right dock, was surprised to see that its header is at the top of the widget instead of the left side.

At first I had thought maybe this was added recently, but then realized this must be from older version of LC.
I would imagine though that in older versions then the CL docked at bottom would have horizontal header as well which would not be desired.
Again was just surprised to see this.

sand1024, 2025-03-29:
@Questions2000 Indeed, such support of different orientation of title bar for Cmd widget was added in one of later version (back in 2016)- as far as I can see from the history, it's related to the docking features. 

Anyway, I suppose I'll make this more uniform and let the docking widgets to support horizontal/vertical title bars. 
That will be included into the next PR from me.  

So it will be something like this: 

[image]

closed 2025-04-22 by Questions2000

sand1024, 2025-04-23:
@Questions2000 
Just a note - icons styling will be operational only if proper shared library with icon engine is included as part of the installation. 

There were recent changes in in build/installation image assembling, and I'm not sure whether that library is included for all platforms. 

The name of library is lc_svgicons.* (different extensions for different platforms) and it should be located in /iconengines directory (near to qsvgicons). 

If this library is missing - well, it's necessary to wait for next 1-2 build where it will be finalized.  Otherwise - it will be possible to style icons.