LibreCAD/LibreCAD #1992: Command Line - Change Zoom, pan or select entity back to Enter Command

Questions2000, 2024-12-31:
Hopefully others will agree with this.
I think this was done because of new info based tools being added, but I noticed that in the command line
`Enter Command:`
had been replaced with 
`Change Zoom, pan or select entity`
when no commands are currently active

I personally really liked the idea of recently changing `Command:` to `Enter Command:` text label because it was very simple and easily understood that there is no command that is currently active/in use.
I feel that the label `Change Zoom, pan or select entity` is a lot of specific information being shown and it makes it feel/appear visually as if a command is currently active, even though there is not.
Suggesting that maybe this can be changed back to `Enter Command:` for better clarity, if possible.

Thank You

sand1024, 2025-01-01:
mmm.... well, that's reasonable point, yet actually there is a small discrepancy in hints. 

Actually, the same message (prompt of the action) is displayed in several places: 

1) cmd widget (prompt)
2) mouse widget (left mouse button hint)
3) + informational cursor (as it was added to the last ApImage). 

So the message "Change zoom... " is actually a command prompt for default action (one which is active if no other actions are active). 

And that's the reason why this text also shown in the command widget. 

For most cases using the same prompt of the action is fine. However, for the situation you've described, I think your point is quite reasonable, so I'll think how to use different texts there.

Questions2000, 2025-01-02:
Ahh thank you for considering to change this back, I appreciate it.

Questions2000, 2025-03-10:
Thank you, I saw that this has been fixed, just not applied yet.
Really looking forward for this update.

I just noticed something
Currently the text reads
`Zoom, pan or select entity`
If you LMB click and hold down this mouse button, the text reads
`Enter Command:`
until you release the button or drag mouse cursor to create selection rectangle.
Once the apply of this fix has been made, this will make sense again.
I cannot verify without the new apply yet, but I assume this hold down button text will remain the same.
This will need to be updated if the following suggestion is agreed upon.

May regret suggesting the following, hard to say if this will look strange now that used to seeing text above the CL field all the time.
I am fine with the "Enter Command:" as requested and leaving it as is.
I was just thinking, I wonder what it would look like if when no command is active, if the space was just blank?
Then text would only be shown when commands are active.
Again I like the "Enter Command:" but was just curious, how this may look/perform.

Thanks again for making this updated

closed 2025-03-21 by Questions2000