LibreCAD/LibreCAD #1994: Named Views Widget - Add ability to sort views manually

Questions2000, 2025-01-01:
### Observed Behavior
I noticed in the "Named Views" widget there is currently no way to change the sort order of view names, new views just get added to the bottom of the list automatically.

### Expected Behavior
Suggesting to add the ability to be able to sort view names manually.

### Suggestions
Here are some ideas of how this can be added.
I can make single posts for these if requested to do so but I will put all in this post for now for clarity because not sure which ones will be considered to be added.  I think they all could be helpful.

`ADD TO VIEW NAMES CONTEXT MENUS`
Each View name has its own context menu when RMB clicking, can add the following items to this context menu
"Move This View Up"
"Move This View Down"

`ADD TO UP/DOWN BUTTONS TO NAMED VIEW WIDGETS TOOLBAR`
Up Arrow Icon for Move Up
Down Arrow Icon for Move Down

`DRAG AND DROP WITH MOUSE"
Allow user the ability to drag named views with LMB click and drag to sort them manually.
This currently is being used to be able to select multiple named views at the same time with mouse, so a decision would need to be made if this is decided to be added.
Would have to remove this LMB drag action from the ability to multi select views and user would just have to use "standard" modifiers used for example in File Managers to select multiple folders/files.
Ctrl + LMB clicks on items to highlight more then one single View name at a time
Shift + LMB (when a single item is already highlighted) apply this when clicking on another item and all of them in between get selected at the same time.
Which would allow this drag and drop sorting suggesting to now use that LMB drag action.

`RELATED POSTS`
#1993

Thank You

sand1024, 2025-01-01:
hm... what is the purpose of such functionality by our opinion? 

Of course, it's possible to support manual/automatic sorting of views, however, it will make sense only of there will be some reasonable amount of views in the drawing (say, more than 10 or so). 

I'm not sure that this is the common case to have lots of named views in the same drawing, but probably I miss something?

Questions2000, 2025-01-02:
Thank you for your reply.
The idea is more about allowing the user the ability to be able to have some way of sorting the views, which is why I also mentioned that other post to allow it this to be done by sequential number/alphabetically.
I think both ideas could be useful, but up to you.
Based on many named views as you mention
but also can be helpful visually to match view names to their position on screen in your drawings.

In a little bit of a rush at the moment, so might not be the best example.
Could also kind of relate to recent post as well #1995 for printing views in paper space.
I was thinking that often if user is going to use named views they would have items (say rectangles) laid out in some kind of a column (vertically) or multiple columns, which would mimic the list of view names in widget.
For example could be a bunch of 8x11 sheet details (rectangles) stacked above each other in a column
Rectangle 1
Rectangle 2
Rectangle 3
Rectangle 4
User might want to have the named views from top to bottom in the same order in widget list as they appear on screen (so easier to remember)
Hard to explain every scenario, but what if a new rectangle wanted to be added above "Rectangle 1"
Its view would be automatically be added to the bottom of the list which would make it out of order, which could be confusing.

The other option mentioned in issue 1993 to sort sequential number/alphabetically could work but then user would have to rename each view to get them to be in order based on screen position.
01 - Rectangle 1
02 - Rectangle 2
03 - Rectangle 3
04 - Rectangle 4
If new rectangle added above the numbering 01 - 04 prefixes would have to be changed to 02-05 so new rectangle 01 can be sorted at the top.

Could also relate if items (rectangles) are horizontal left to right as well, just that columns example was easier to illustrate because lists of names are already in column format top to bottom.
Also easier to move names in a list around rather than having to create new views over and over to get them in the right order visually.

Sorry, hopefully some of this makes some sense :)
Thank you again for replying / interest in the topic.

Questions2000, 2025-01-03:
MANY NAMED VIEWS EXAMPLE
As mentioned "detail" drawings are common and projects can include many of them all in one drawing.  If user had a drawing with 20-40 detail drawings and wanted to have a named view for each of them to navigate through them, which is similar to the example you mentioned, where many named views would be present.

ANOTHER SIMPLE SCREEN POSITION BASED EXAMPLE
Again if drawing file contained many "detail" drawings and user wanted to match the position on screen of the detail drawings to the order they appear in the named view list so easier to keep track of.  If user had a named view for each detail and then wanted a new named view containing all of them, as a group, they might want that group name to be at the top above the indiividual detail named views, but all new named views are automatically added at the end of the list
Detail 1
Detail 2
Detail 3
Detail 4
Group 1 (view of Detail 1-4, want this named view of the group to appear at top of the list above Detail 1 named view.