LibreCAD/LibreCAD #1995: Named Views Widget - Ability to save views in Print Preview

Questions2000, 2025-01-01:
Suggesting to add the ability to save Named Views when in Print Preview mode.
Terminology can get a little confusing on what these modes are called, I apologize not sure if this is already possible?
I noticed that in Named Views Widget if you hover over View names it shows information tags about each view.
Tags mention
`In Paperspace: No`
Is this already possible? if not as mentioned making a suggestion for this.
(I know that "Modelspace" and "Paperspace" has been mentioned for LibreCAD3, was going to make a post about reconsidering this to be added to LibreCAD2 but have been trying to hold off doing so because I do not think it will be considered to be added)

Thank You

sand1024, 2025-01-01:
Well, this is planned to be added- yet before adding that feature, a more complete support of paper space should be implemented - and only after this it will be reasonable. 

So this will be added later in some future, yet probably not in a couple of nearest updates.

Questions2000, 2025-01-01:
Okay, thank you was a little confused, was not sure if I was unaware of something.
Good to hear that this is planned in the future.
So I will leave this post open.
Thank You