LibreCAD/LibreCAD #1997: Change name of some of the Zoom based tools

Questions2000, 2025-01-02:
Not important, I ran across a few things to mention about "Window Zoom" so I thought I would just mention this as well.
Suggesting to change names of the following Zoom based tools
`Auto Zoom` - `Zoom Auto`
`Window Zoom` - `Zoom Window`
This way each of the Zoom tools would start with the word “Zoom”

dxli, 2025-01-27:
It's already so:

za	zoomauto
zp	zoompan
zv	zoomprevious
zw	zoomwindow

Questions2000, 2025-05-10:
I was basing this on the hover tag labels that show up when you hold mouse cursor over the zoom based icons.
Most of them start with "Zoom"
I can close this if not going to be changed.
You guys let me know.

dxli, 2025-05-11:
I see the issue is to decide between:

"Zoom Auto"  and "Auto Zoom"

Questions2000, 2025-05-14:
Yes and 
`Window Zoom` - `Zoom Window`
Making these hover tag labels start with "Zoom" prefix like the other Zoom based tool hover labels do.

closed 2026-10-10 by Questions2000