LibreCAD/LibreCAD #2000: Opening Current Drawing Preferences Dialog crashes the application

sclebo05, 2025-01-02:
### Expected behavior
Current Drawing Preferences Dialog should be displayed when selected from the Options menu.

### Observed behavior
Program Crashes with the following output captured from CLI: 
$ librecad 
RS_DEBUG::setLevel(3)
RS_DEBUG: Critical
RS_DEBUG: Errors
RS_DEBUG: Warnings
QLayout: Attempting to add QLayout "" to QG_LibraryWidget "Library", which already has a layout
QG_LayerWidget::update: nullptr lastLayer
*** stack smashing detected ***: terminated
Aborted (core dumped)

### Steps to reproduce or sample file
1. Open LibreCAD
2. Select from the top menu Options -> Current Drawing Preferences
3. Application crashes with above output.

This appears to happen with a saved or unsaved file.

### Operating System and LibreCAD version info
EndaevourOS, an Arch derivative - current as of 1/2/2024
LibreCAD package version 2.2.0.2-2e
I can supply any library versions if it is helpful.

sand1024, 2025-01-02:
@sclebo05 
Could you please check whether crash occurs on ANY dxf file or just on some specific ones? 

If it occur only on some specific dxf - could you please share that file? 

If crash is on for any drawing - try to use fresher build, either from 2.2.1 or latest 2.2.2 tag
You can find them there: 
2.2.1 - https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.1_rc-latest
2.2.2 - https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.2_alpha-latest

sclebo05, 2025-01-03:
Compiled 2.2.2_alpha.latest.r229.g7528d8184-1 from the Arch User Repo and confirmed that the issue was no longer present.

Of note - the website download for Arch leads to a 404 error, and the GitHub page shows the last release as 2.2.0.2. When there's a new release, the Arch package maintainer will no doubt ship it.

closed 2025-01-03 by sclebo05