LibreCAD/LibreCAD #2002: SEGFAULT when using "Trim two"

Zmuogs, 2025-01-03:
Hello. I'm trying to use the "Trim two" function on two segments (test file: [trim_two__segfault__02.dxf.zip](https://github.com/user-attachments/files/18299506/trim_two__segfault__02.dxf.zip)).
The drawing contains two lines:
- activate the "Trim two" function,
- click on the horizontal line (point 1),
- then move the cursor to the vertical line (arrow 2) -> the LibreCad crashes with the SEGFAULT error.

[image]
There is a console output [trim_two__segfault__debug_6_output.txt](https://github.com/user-attachments/files/18299616/trim_two__segfault__debug_6_output.txt). The command in the console was "librecad --debug 6 trim_two__segfault__02.dxf".

LibreCad version: 2.2.2_alpha1-231-g7528d8184<br/>Compiler: GNU GCC 14.2.1<br/>Compiled on: Jan  2 2025<br/>Qt Version: 6.8.1<br/>Boost Version: 1.86.0
Operating System: Arch Linux 
KDE Plasma Version: 6.2.5
KDE Frameworks Version: 6.9.0
Qt Version: 6.8.1
Kernel Version: 6.6.68-1-lts (64-bit)
Graphics Platform: X11
Processors: 16 × AMD Ryzen 7 1700X Eight-Core Processor
Memory: 62.7 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1080 Ti/PCIe/SSE2

dxli, 2025-01-05:
Does this happen to 2.2.1?

Zmuogs, 2025-01-05:
I tried "LibreCAD-v2.2.1-x86_64.AppImage" and this bug does not occur there.

sand1024, 2025-01-05:
Oh, I'll take a look on this.

sand1024, 2025-01-05:
will be fixed in next update

Questions2000, 2025-01-23:
I thought I would give this a test to help out if I could.
I tested this out in the latest Linux Appimage available
`LibreCAD-2.2.2_alpha1-243-gb10a28bb.AppImage`
and it does not appear to be fixed.

Couple things I noticed.
-If the vertical line goes beyond the horizontal line to form a "t" or "+" shape, then this issue does not occur
-Opposite way, if the vertical line ends before it gets to the horizontal line, this issue also does not occur
-This only happens with exactly how the original user mentions to generate the issue, where the vertical line ends right at the horizontal line.

Questions2000, 2025-02-01:
This issue appears to be fixed, but now seems to generate a new issue that can be found here if anyone is interested
https://github.com/LibreCAD/LibreCAD/issues/2029

closed 2026-09-26 by sand1024