LibreCAD/LibreCAD #2004: Using QSS Style Sheet causes new windows to open per drawing upon closing LibreCAD

Questions2000, 2025-01-03:
BUG REPORT
I still feel that this bug fix is important, so I decided to close my other post and make this new one regarding this issue, hopefully this will be worth the time spent to do so.
For anyone not familiar with what QSS is used for, I will add some info in comment below that maybe could be helpful.

### Expected Behavior
For any drawings opened within LC to ALL close when LC application itself is closed.

Same as a web browser, if you have multiple tabs open, when you close the web browser, all the tabs opened within it all close at the same time.

### Observed Behavior
If you have a QSS Style Sheet linked to your LC application, when you close LC application, LC application window closes, but then ALL opened drawing tabs within LC application end up opening in their own separate windows and have to also be closed.
So for example if you have 5 drawings opened and you go to close LC application, 5 new windows all open and you have to close each of the 5 windows one at a time.

### Steps To Reproduce
Long set of instructions so made in its own comment below this post.

### Operating System and LibreCAD Version
-Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-227-g05bc6ccb.AppImage
-Slackware (Other user mentioned)
-Windows OS

So it appears to not be OS related, but something with either LC or Qt toolkit or both.

### Related Posts
I think my post above covers most of the information about this issue and will hopefully be the one to use, but here are some links of posts that mention this issue as well, if further information is required.
#1557 - Old post that was CLOSED
#1794 - My original post about this issue that I CLOSED
#1810 - Newer post a different user made after my post (I do not have the authority to CLOSE other users posts, I just asked user to do so, unless someone else has ability to close it?)

Questions2000, 2025-01-03:
### Steps To Reproduce
-Create a text files called `test.qss`

-In this text file add the following line and save the file
`QMenu { font-size: 24px; }`
(Note that if this .qss file is left blank then this bug does not occur, it only occurs when there is QSS based data saved within that .qss file)

-Launch LC and maximize the window
-Click on `Options` in your Menu Bar and click on `Widget Options` in drop down
-At the top of the `Widget Options` dialog you can leave the `Style` box unchecked and set `Windows`  in the drop down to its right.
-In that same `Widget Options` dialog, next click on the `Style Sheet` button and navigate to that `test.qss` file you saved in step one.
-The path and test.qss filename should appear in the field to the right of the 'Style Sheet" button
-Then click `Save` button at bottom of the `Widget Options` dialog

-To verify that this QSS worked, click on any category in your Menu Bar and you will notice that now the menu font is larger

-Now in blank drawing that opens when you launched LC, draw a circle, save drawing, leave drawing open.
-While that circle drawing is still opened, create a new drawing tab and draw a rectangle, save drawing, leave drawing open.
-While those 2 drawings are still opened, now close LC application itself.
Closing method does not matter can use Alt+F4, Ctrl+Q or LC application “X” icon at top right corner

-You will notice that when you go to close LC application, LC itself closes, but 2 new maximized windows open which are the drawing with the circle and drawing with the rectangle.
Both of those new windows have no header/title bar, no toolbars, just shows the drawing pane and vertical & horizontal scroll bars.
This is the bug I am looking to have corrected.  All drawings should close when LC application itself is closed.

### Steps To Revert Back To Previous State
If you do the following the issue goes away, goes back to previous state
-Open `Widget Options` dialog again
-Clear the path & test.qss filename from the `Style Sheet` field
-Click on `Save` button
-The multiple windows will happen once more
-Next time you launch LC then the issue is gone

Questions2000, 2025-01-03:
### QSS Info
For those who do not know “QSS” is short for "QT Style Sheet"
It is the QT toolkits version of “CSS” which is used for example by “GTK” toolkit.
QSS files allow user to customize the appearance of QT based applications, such as LibreCAD.
If elements within LC are coded to support QSS, user can customize things such as font, font size, font colors, background colors, etc.
There are quite a few QSS Style Sheet related items built within LC itself, for example
`Menu Bar - Options` contains two items regarding this
-Widgets Options (contains QSS settings)
-Reload Style Sheet

Many items within the LC GUI are coded to support QSS modifications.
There are also QSS examples provided in LC manual which can be found here
https://docs.librecad.org/en/latest/ref/customize.html

So QSS is clearly something that LC was designed to support and there are no other ways to modify specific items in LC GUI without it, so this bug is an issue that hopefully can be addressed.
Although QSS still works, this bug will cause users to have to make the decision if they want to put up with an inconvenient bug in order to alter the appearance of their LC application.

Questions2000, 2025-01-03:
@sand1024
Sorry to tag you about this, not sure if you would be interested, but reading the “related posts” mentioned above, other developers have not been able to find anything that could be causing this issue, you are the only developer that I have not seen a response in any of those posts and maybe can add a different perspective to narrowing down the cause of this issue and if it is even possible to fix.
The user who originally posted issue #1557 was created in Aug 11, 2022 so this issue has been around for a while.
Again sorry
Thank You

Questions2000, 2025-01-18:
If anyone else is interested, I finally came up with a method to testing this in the terminal with -d6 option.

LC IS OPENED
(THIS IS THE BOTTOM OF THE INFO SHOWN, NOT RELEVANT)
`LC_LayerTreeWidget::setLayerList()
QG_BlockWidget::update()
QG_BlockWidget::activateBlock()
QG_BlockWidget::update() done
QC_ApplicationWindow::slotFileNewHelper: load Template: OK
QG_LayerWidget::update() begin
QG_LayerWidget::update(): OK
QC_ApplicationWindow::slotFileNewHelper: update coordinate widget
QC_ApplicationWindow::slotViewGrid()
RS_VariableDict::addVariable()
QC_ApplicationWindow::slotViewGrid() OK
QC_ApplicationWindow::slotFileNewHelper() OK
QC_ApplicationWindow::slotFileNewNew() OK
main: entering Qt event loop`

u2fly, 2025-04-18:
@dxli @sand1024 

Still happens in both continious 2.2.1_rc and 2.2.2_alpha

Reproduce
1. Launch LibreCAD
2. Set "Widget > Style Sheet"
3. Close LibreCAD
4. Default `unnamed document 1` window untached from main window, while main window closed

If remove style sheet and restart there is no issue.

sand1024, 2025-04-18:
@u2fly 
Yes, that's was also changed yet not committed yet. Should be available in nearest couple of days.

Rehab42, 2025-08-28:
Version: v2.2.1.2
Compiler: GNU GCC 11.4.0
Compiled on: Jul 13 2025
Qt Version: 5.15.2
Boost Version: 1.74.0
System: Debian GNU/Linux 12 (bookworm)

This is still an issue. will it bee in the next release?

sand1024, 2025-08-29:
@Rehab42 
I'm not sure this fix included in 2.2.1 branch (production), my comment was for 2.2.2-alpha. 
You may check it there - https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.2_alpha-latest

Rehab42, 2025-08-29:
> [@Rehab42](https://github.com/Rehab42) I'm not sure this fix included in 2.2.1 branch (production), my comment was for 2.2.2-alpha. You may check it there - https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.2_alpha-latest

Thank you @sand1024 I must have missed that thanks!

Questions2000, 2025-09-09:
@sand1024 
Tested this out in the following version and this works now.  Guess it was not an issue with QSS after all.
Wow, thank you very much for making this update, I did not think this one was going to be updated.

This post was a double by a different user so I cannot close it personally.
https://github.com/LibreCAD/LibreCAD/issues/1810

closed 2025-09-09 by Questions2000