LibreCAD/LibreCAD #2006: Diameter symbol (ø) can't be added with buton

bstajcar, 2025-01-05:
### Expected behavior
 Diameter symbol (ø<>) should be possible to add with Label button in Linear dimension Properties dialog box

### Observed behavior
Clicking to Label button does nothing
(The desired Label can be added manually)

### Steps to reproduce or sample file

### Operating System and LibreCAD version info
Windows '7, LibreCAD Version: v2.2.1

dongxuli-concord-design, 2025-01-05:
Just tested the official mac build, and the issue cannot be reproduced.

Could be a windows issue though. Just in case, please test both 32-bit and 64-bit builds. 2.2.1 is expected to be the last version with official 32-bit support though.

RockVicedomini, 2025-03-22:
I can confirm the same issue with the latest release candidate appimage for Linux: 
LibreCAD-v2.2.1.1-9-g5ad9b999-x86_64.AppImage 

With the older version 2.2.0.2, clicking the ø button in the 'Linear Dimension' window  add the symbol to the dimension

dxli, 2025-03-22:
@RockVicedomini 

Can you edit a dimension and manually make the label  '(ø<>)'?

dxli, 2025-03-23:
@RockVicedomini 

Just tested LibreCAD-v2.2.1.1-9-g5ad9b999-x86_64.AppImage, and I couldn't reproduce this issue.

It could be a font issue for ø:

Please verify your default font path ( Edit -> Application Preferences > Paths ).

Also, try to delete the default font path and restart librecad, so the appimage can use the font files within appimage.

RockVicedomini, 2025-03-23:
> [@RockVicedomini](https://github.com/RockVicedomini)
> 
> Can you edit a dimension and manually make the label '(ø<>)'?

Yes, I can edit it manually with no problem. It fails to add the symbol only when I press the button in front of the label field

dxli, 2025-03-23:
@RockVicedomini 

I still couldn't reproduce this issue with app image.

RockVicedomini, 2025-03-23:
> [@RockVicedomini](https://github.com/RockVicedomini)
>
> Just tested LibreCAD-v2.2.1.1-9-g5ad9b999-x86_64.AppImage, and I couldn't reproduce this issue.
>
> It could be a font issue for ø:
>
> Please verify [...]

I did not specify any font path. The 'fonts' field is empty in the Application Preferences window

My OS info:

Operating System: KDE neon 6.3
KDE Plasma Version: 6.3.3
KDE Frameworks Version: 6.12.0
Qt Version: 6.8.2
Kernel Version: 6.8.0-51-lowlatency (64-bit)
Graphics Platform: X11
Processors: 4 × Intel® Core™ i3-3240 CPU @ 3.40GHz
Memory: 7.7 GiB of RAM
Graphics Processor 1: NVIDIA GeForce GTX 750 Ti
Graphics Processor 2: llvmpipe

Librecad Appimage info:

Version: v2.2.1.1-9-g5ad9b999
Compiler: GNU GCC 9.4.0
Compiled on: Mar 15 2025
Qt Version: 5.15.2
Boost Version: 1.71.0
System: KDE neon 6.3