LibreCAD/LibreCAD #2008: [Plugins] Bulk Drawing

widifadi, 2025-01-06:
Hello, I wonder if there's a way to perform bulk drawing through plugin. I'm currently developing a plugin that deals with drawing a lot (hundreds to even thousands) of objects, and experiencing app freeze since there's no way to put the drawing task outside the main thread. 

Also I notice that entities drawing that stored in the Plug_Entity container are drawn much faster than using "add [object]" methods that available on plugin_interface. I wonder why is that and if it is possible to store the data directly to Plug_Entity container for alternative to faster drawing?

sand1024, 2025-01-06:
hm... not sure I understand this. 

>app freeze since there's no way to put the drawing task outside the main thread
by "drawing task" - you mean creation of entities and insertion them into drawing? 

As for drawing speed - as soon as entity is created and is put into the drawing, there should be in general no difference in drawing speed.  

I suspect that it might be, that your code may use some logic, other than expected one, which is as follows (roughly):  

1) the plugin is invoked:
2) it performs some calculations and creates the set of entities
3) entities inserted into the drawing
4) plugin ends

As soon as entities are created and added to the drawing, they are exactly the same to entities created by the user, and it's up to the engine to decide when to draw them.

widifadi, 2025-01-07:
So to give more context, this is the example code I used in my plugin: 
```C++
std::vector<qulonglong> PainterService::drawLine(QList<QPointF> pts, QString layer)
{
    doc->setLayer(layer);
    if (pts.empty()) {
        qDebug() << "No need to draw the line, points are empty";
        return {};
    }

    std::vector<QPointF> points;
    for (const auto& pt3d : pts) {
        double x = pt3d.x();
        double y = pt3d.y();

        points.emplace_back(QPointF(x, y));
    }
    return doc->addLinesReturn(points);
}
```
The code above is what I meant for the drawing task. When I tried to put this process in concurrent, apparently it need to be invoked back to the main thread. As a result, if I want to perform drawing of a bunch of lines it will iterate and call this function for 100 times not in parallel. In my case I'm not only drawing lines, but might as well other entities such as text and else, which ended up taking a lot of time and may cause the app freeze I mentioned. 

Which comes down to the idea maybe having a bulk drawing function in plugin_interface might solve my problem. The first approach for bulk drawing that I have in mind is passing this data set (objects to be drawn) to plugin_interface and have a function that handle this (parsing, calculations) to produce the set entities that need to be inserted to the drawing. But I think it will end up the same to my current approach. 

The other idea that comes to mind is to put directly the data that want to be drawn directly to the Plug_Entity. I found that using the following code, eventhough the core process is different (one is producing new set of entities and this one is basically just make a copy of Plug_entity), the drawing process of entities that already stored in Plug_entity takes in a matter of seconds. 
```C++
QList<Plug_Entity *> CopyBasePoint::copySelected()
{
    QList<Plug_Entity *> ents;
    bool yes = doc->selectByWindow(&ents, "Select objects to be copied: ");
    if (!yes || ents.empty()) {
        emit sendNotification("No object selected!");
        return QList<Plug_Entity *>();
    }

    m_ents = ents;
    emit sendNotification("Entities Stored.");

    return m_ents;
}
```