LibreCAD/LibreCAD #2010: add number dimension when i'm adding line, polyline...

manuelsongokuh, 2025-01-08:
hi

i fisrt time to use librecad, i add new polylines but i wish add my number dimension, but i can't add my number.. how?

maybe librecad has not this feature "adding dimension while i adding..."

i use Rhinoceros 3d, has this feature i add any my dimension each times, this is easy and fast to create lines, polylines..but librecad not..

manuelsongokuh, 2025-01-08:
see example by QCAD has this feature ADDING number dimension.. this is faster to create and easy...

https://github.com/user-attachments/assets/39741204-39a7-4647-99d6-16d1a1278694

gerardbm, 2025-01-11:
+1 for this. Allplan also uses this, where it's faster to add lines with different distance sequentially.

sand1024, 2025-03-29:
@manuelsongokuh 
@gerardbm 

Well, it is possible to enter length of line segment, yet in a bit different way. There are several ways how to do this. 

1) Length of line is supported by some Line command (angle, horizontal, vertical)
2) the simplest way to specify length - is enter second point of line segment in RELATIVE-POLAR coordinates using Cmd widget (command line). 

so, line starting from 0.0, with length 10 and angle 45 will be drawn by the following set of commands:
line
0,0 
@10<45

3) if you need to draw vertical or horizontal line - there is even simpler way. 
Specify first point of line, enter the distance and then in the command widget command line press 

CTRL+UP/DOWN/LEFT/RIGHT (one of, for sure) - thus you'll specify the direction to which previously entered distance should be applied.  

both 2 and 3 above a quite uniform and may be used in any place where the action expects for coordinate. 

---
As for adding length as parameter to UI - well, it might be a nice addition, yet it's hardly will let entering commands faster than via command line.  However, we may add this too.

sand1024, 2026-09-26:
I suppose this is supported by relative input assistant.

closed 2026-09-26 by sand1024