LibreCAD/LibreCAD #2011: Crash when change paper size

IMurzich, 2025-01-08:
### Expected behavior
Paper size changed successfully

### Observed behavior
LibreCAD crashed with memory dump

### Steps to reproduce or sample file
1. Unpack and open [file.tar.gz](https://github.com/user-attachments/files/18346375/file.tar.gz)
2. Options -> Current Drawing Properties -> Paper -> Format -> Change to A4
3. Ok

### Operating System and LibreCAD version info
Version: 2.2.2.3-alpha (build from git commit 5899a9cf6e92d1cd7514cbdcaf09c407c94eb92c)
Compiler: GNU GCC 13.3.0
Compiled on: Jan  8 2025
Qt Version: 6.8.1
Boost Version: 1.83.0
System: KDE neon 6.2

dxli, 2025-01-14:
@sand1024 

I cannot reproduce this issue. Could you have a look?

sand1024, 2025-03-29:
@dxli 
@IMurzich 
I also can't reproduce this on this file and on current sources (yet I'm under win10, but as far as I can see, it's hardly possible that there could be something that is OS-specific).