LibreCAD/LibreCAD #2012: changing tool resets "lock relative zero"

flrl, 2025-01-12:
Just updated from 2.2.0 to 2.2.1, and noticed that the "lock relative zero" state no longer sticks.  After a little experimenting, it seems to be getting reset when changing the selected tool.

### Expected behavior

1. Set relative zero position to somewhere in the document
2. Lock relative zero position
3. Select any tool (e.g. 2 Points line)
4. Draw
5. Relative zero position is still at the position you set in step 1
6. Lock relative zero position is still set

### Observed behavior

1. Set relative zero position to somewhere in the document
2. Lock relative zero position
3. Select any tool (e.g. 2 Points line)
4. **Lock relative zero position has become unset**
5. Draw
7. **Relative zero position updates as you draw**

### Operating System and LibreCAD version info

Version: v2.2.1
Compiler: GNU GCC 12.2.0
Compiled on: Jan  4 2025
Qt Version: 5.15.2
Boost Version: 1.87.0
System: Windows 10 Version 2009

sand1024, 2025-02-02:
@dxli I'll commit fix with next Pull Request

sand1024, 2025-03-29:
fixed in #2032

dxli, 2025-03-29:
@sand1024 

Please keep PRs small. When some mysterious regression happens, we can always do a quick bisection to find the first offender commit.

This would be less helpful, if PRs are huge.

Aside from that, smaller PRs could be reviewed with proof of correctness sometimes.

Since we don't have a full time maintenance team, proof of correctness and easy to understand changes are essential to prevent to code getting obsolete once the original developer leaves the project.

sand1024, 2025-03-29:
@dxli 
Yes, I actually try to do this - yet unfortunately sometimes refactoring like rename affects lots of files... Or some feature just affects too many places.

sand1024, 2025-04-23:
@flrl 
this is fixed in PR #2114 for 2.2.2 alpha

dxli, 2026-09-14:
Re-checked on the current branches by driving the application window. The lock now holds when picking tools, drawing, pressing Esc and switching drawings.

- **2.2.1:** fixed by e0f5dd6b0, first in v2.2.1.1. That change left tool buttons checked after their action ended. #2791 (72731a393) fixed that on the 2.2.1 branch, and it will be in the next 2.2.1 release.
- **master:** fixed by #2032 and #2114.

One keyboard-only gap was left on master. After Set Relative Zero was given a typed coordinate, the action stayed current until the mouse moved, so the next typed coordinate could move the locked relative zero. #2862 fixes that, along with the lock button's tooltip after switching drawings.

Closing as fixed.

closed 2026-09-14 by dxli