LibreCAD/LibreCAD #2013: "Cancel"-button closes drawing without saving instead of aborting the close operation

ElDonRicko, 2025-01-12:
When attempting to close a drawing without previously saving and selecting "Cancel" or "X" (Top-right corner) the drawing gets closed without being saved.

### Expected behavior
The Close operation should be stopped and the Drawing should remain open when selecting "Cancel" or "X"

### Observed behavior
The Drawing gets closed without being saved, as if selecting "Discard".

### Steps to reproduce or sample file
Close any Drawing without saving and choose "Cancel" or "X"

### Video
https://github.com/user-attachments/assets/2ad9340f-7c89-4cab-857b-8ba70a4f78d9

### Operating System and LibreCAD version info
Version: 2.2.1_rc4
Compiler: GNU GCC 14.2.1
Erstellt am: Jan  7 2025
Qt Version: 5.15.16
Boost Version: 1.86.0
System: Version: 2.2.1_rc4
Compiler: GNU GCC 14.2.1
Erstellt am: Jan  7 2025
Qt Version: 5.15.16
Boost Version: 1.86.0
System: Arch Linux x86_64

sand1024, 2025-04-23:
@ElDonRicko 
Could you please check this in 2.2.2 alpha branch? The logic for closing is changed there.

ElDonRicko, 2025-04-26:
@sand1024 
Everything now works as expected, except when I try to close only the tab of the current drawing.
"Yes" and "No" work correctly, but selecting "Cancel" or clicking the "X" (to close the pop-up) creates a new temporary save file named "#Name.dxf".
If I then attempt to close and cancel again, the application saves "#Name.dxf" as "Name.dxf" and deletes the temporary "#Name.dxf" file.

When using the top-right "X" button to close the entire application, "Cancel" and "X" on the pop-up close the pop-up correctly every time, which is the expected behavior.

I hope my description is understandable, as I am not a programmer.

sand1024, 2025-04-26:
@ElDonRicko 
Oh, thank you for the detailed description. Yes, it seems that the is some glitch with "cancel" logic. 

As for #name.dxf file - it's a backup file, it's created from time to time as well as auto-save version of the file. Yet anyway, I'll check again.

sand1024, 2026-09-26:
fixed

closed 2026-09-26 by sand1024