LibreCAD/LibreCAD #2014: Aliases file librecad.alias location on Linux is different from documentation

emulti, 2025-01-13:
### Expected behavior
librecad.alias is read from `.local/share/data/LibreCAD/librecad.alias` and all edited aliases are loaded.

### Observed behavior
librecad.alias is read from` .local/share/LibreCad/LibreCAD/librecad.alias`
Tested aliases are not loaded

### Steps to reproduce or sample file
add a simple tab-separated librecad.alias file to `.local/share/LibreCad/LibreCAD/librecad.alias` OR `.local/share/data/LibreCAD/librecad.aliias` :
```
text    dt      # draw text
modexplode      x       # explode
zoompan z       # zoom pan
zoomprevious    zp      # zoom previous
```

run librecad with -d6 flag in terminal:
...
```
QC_ApplicationWindow::QC_ApplicationWindow: init settings
QC_ApplicationWindow::initSettings()
QC_ApplicationWindow::slotFileAutoSave(): begin
appData: /home/chris/.local/share/LibreCAD/LibreCAD

getDirectoryList(): line 601: dir=/usr/bin

```
....

in cmd/rs_commands.cpp l199 the location is given as
`* In OS_LINUX "/home/<user>/.local/share/data/LibreCAD/librecad.alias"`

The edited aliases are not loaded. The standard aliases continue to work. 
This happens if either of the locations in .local/share are used.
(I checked for conflict with commands in lc_commandItems.h

### Operating System and LibreCAD version info
Arch Linux with 6.6.69-1-lts kernel, installed from repo
Version: 2.2.1_rc4
Compiler: GNU GCC 14.2.1
Compiled on: Jan  7 2025
Qt Version: 5.15.16
Boost Version: 1.86.0

lordofbikes, 2025-01-14:
The reference in `rs_commands.cpp` is just a comment and might be outdated.
There is no OS specific code in LibreCAD source, basically the location is red by a Qt function which requests the location from the OS.
https://github.com/LibreCAD/LibreCAD/blob/82d80a0103ec63ca834e5a96ede0be8a8054a002/librecad/src/lib/engine/rs_system.cpp#L524-L534
Thus there is not _the_ `OS_LINUX` location, it depends on the Linux flavour which location is delivered finally.
On my Ubuntu 24.04 Linux the location is `/home/<user>/.local/share/LiberCAD/LibreCAD`.

So probably there is already a `librecad.alias` on your Arch Linux which you should find and modify to apply your changes.

emulti, 2025-01-14:
Yes, the location used is `~/local/share/LibreCAD/LibreCAD/librecad.alias` on Arch. 
Maybe someone with commit access can correct the comment in `rs_commands.cpp` l199 ?

However, modified aliases at this location don't appear to be read:

with this tab-separated list in librecad.alias:

```
text    dt
modexplode      x
zoompan z
zoomprevious    zp
```

This is the resullt (no remapped aliases are reported in the debug log):
```
x
Unknown command: x
dt
Unknown command: dt
z
Unknown command: z
Accepted keycode: zp
```

But the command loaded with 'zp' is zoompan, the standard alias, not the remapped 'z':

```
QG_DialogFactory::commandMessage
QG_DialogFactory::commandMessage: OK
QG_ActionHandler::setCurrentAction()
QC_ActionHandler::setCurrentAction: view = 0x64188eb4df80, document = 0x64188f1a4610
RS_ActionInterface::RS_ActionInterface: Setting up action: "Zoom Panning"
RS_ActionInterface::RS_ActionInterface: Setting up action: "Zoom Panning": OK
RS_EventHandler::setCurrentAction
RS_EventHandler::setCurrentAction Zoom Panning
```

The reason I am trying to set up these aliases is to make them match to the default set in Autocad. A few more changes are necessary, but when they didn't work I decided to test with this minimal set. Unfortunately, they don't work.

I worked through `cmd/lc_commandItems.h` and manually extracted the attached list of standard aliases. (There are some in there which are not in the User Manual on the website). I wanted to be sure that that if I add Autocad-style aliases they don't conflict with any others.  The list is sorted by short-command.
[librecad.alias.std.txt](https://github.com/user-attachments/files/18408246/librecad.alias.std.txt)

Maybe this file could be used as the standard librecad.alias template created by the program, instead of a blank one. Then there would be less chance of users creating conflicting aliases when adding their own. 

However, with 2.2.1_rc4 at least, custom aliases don't appear to be working.

emulti, 2025-01-14:
I built 2.2.2_alpha from source. 

Version: 2.2.2_alpha1-236-g82d80a010<br/>Compiler: GNU GCC 14.2.1<br/>Compiled on: Jan 14 2025<br/>Qt Version: 6.8.1<br/>Boost Version: 1.86.0
System: Arch Linux

librecad.alias was: 

```
text    dt
modexplode      x
zoompan z
zoomprevious    zp
modstretch      s
```

`librecad -d6` gives some more debug info: 

```
appData: /home/chris/.local/share/LibreCAD/LibreCAD

updateAlias(): Command alias file: /home/chris/.local/share/LibreCAD/LibreCAD/librecad.alias
updateAlias(): invalid alias, command not found: text   dt
updateAlias(): requesting alias(modexplode) for unknown command: x
updateAlias(): invalid alias, command not found: zoompan        z
updateAlias(): cannot overwrite existing alias: zoomprevious=zoompan: with zoomprevious=zp
updateAlias(): invalid alias, command not found: modstretch     s
updateAlias(): done
```

Am I doing something wrong in constructing the librecad.alias file?
The format I used is
```
command1<tab><alias>
command2<tab><alias>
...
```

emulti, 2025-01-14:
It turns out that the alias file format (LibreCAD  2.2.2_alpha)  is:
`<alias><tab><command>`

See `rs_commands.cpp` line 234.
```
            const QString& alias = txtList[0];
            const QString& cmd = txtList[1];
```

librecad.alias: 
```
dt      text
x       modexplode
z       zoompan
zp      zoomprevious
s       modstretch
```

Result:

```
updateAlias(): begin                                                                                                                                                                                                            
appData: /home/chris/.local/share/LibreCAD/LibreCAD                                                                                                                                                                                                                                                                            

updateAlias(): Command alias file: /home/chris/.local/share/LibreCAD/LibreCAD/librecad.alias                                                                                                                                                                                                                                   
updateAlias(): cannot overwrite existing alias: zp=zoomprevious: with zp=zoomprevious                                                                                                                                                                                                                                          
adding command alias: dt        text                                                                                                                                                                                                                                                            

adding command alias: s modstretch                                                                                                                                                                                                                                                      

adding command alias: x modexplode                                                                                                                                                                                                                                                      

adding command alias: z zoompan                                                                                                                                                                                                                                             

updateAlias(): done                                   
```
But it is not yet clear to me why 'cannot overwrite existing alias: zp=zoomprevious: with zp=zoomprevious' occurs, because the existing alias is zp=zoompan which has previously been redefined.

emulti, 2025-01-15:
thanks for the response!
My mistake to get the librecad.aliases file format the wrong way around...
Overriding existing aliases would be very convenient, especially for former Autocad users who have learned a different set of shortcuts.
Looking forward to this functionality in 2.2.2. Happy to test when ready.