LibreCAD/LibreCAD #2015: Symantec rise the Uninstaller as Security Risk

mtbadi39, 2025-01-14:
### Expected behavior

### Observed behavior
Symantec rise the uninstaller as Security Risk
### Steps to reproduce or sample file

### Operating System and LibreCAD version info
LibreCAD-Installer-2.2.0.2.exe
Windows 10 Professionnel 19045.4780

[image]

dxli, 2025-01-14:
Please do not download from source forge.

Use GitHub instead.

lordofbikes, 2025-01-14:
LibreCAD uses [NSIS](https://nsis.sourceforge.io/Main_Page) for Windows packaging and the addressed `uninstall.exe` is the same, on SourceForge and on GitHub.
So using GitHub packages will probably lead to the same result.

Anyhow, as the report references an early 2.2.1-alpha release, you may want to switch to the brand new [2.2.1](https://github.com/LibreCAD/LibreCAD/releases/tag/v2.2.1) release. The used NSIS version might be newer and solve the Symantec issue.

If the issue persists with 2.2.1 release, the recommend action is to report the NSIS `uninstall.exe` as a false positive to Symantec directly.