LibreCAD/LibreCAD #2018: Alias File - Add path for this file in Application Preferences

Questions2000, 2025-01-17:
### Expected Behavior
To have all settings files be in the same location, for example in Linux `$HOME/.config/LibreCAD`
Suggesting (if it is possible) to add the ability for user to be able to set the path location to the `librecad.alias` file.
Can be added in Application Preferences - Path tab.

### Observed Behavior
The main three settings files created/used by LibreCAD are....
librecad.alias = command line aliases
LibreCAD.conf = config file
default.lcs = shortcut mappings

In Linux these files are located in the following locations....
librecad.alias = `$HOME/.local/share/LibreCAD/LibreCAD/ (this path cannot be changed by user)
LibreCAD.conf = `$HOME/.config/LibreCAD` (this path also cannot be choosen by user, but for me this is the desired location)
default.lcs = `$HOME/` (can already change in Application Preferences - Path tab)

User cannot change the path location of the `librecad.alias` file.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-234-gdb4ff4d9.AppImage

Questions2000, 2025-03-30:
I was wondering if this request would be considered?
Does anyone know if this is possible, or if there is a specific reason why it was decided to put this `librecad.alias` file in this folder
`$HOME/.local/share/LibreCAD/LibreCAD/`?

Thank You

dxli, 2025-03-30:
@Questions2000 

For settings, we follow the standard locations: 

https://doc.qt.io/qt-6/qstandardpaths.html

Those locations may conceptually be changed in the future.

sand1024, 2025-04-23:
@Questions2000 
please test in 2.2.2, this was changed in one of previous updates and now alias file should be stored in Other Settings Directory location, if its' specified in settings. If this path is empty in Paths settings, application data dir is used, as before. 

If it works as expected - please close the issue.

Questions2000, 2025-05-20:
This was updated in post 2103
Thank You

closed 2025-05-20 by Questions2000