LibreCAD/LibreCAD #2021: Version 2.2.1 - Error on file open: Windows 10 x64, Visual C++ Runtime error, Assertion Failed

veperr, 2025-01-20:
I've had this happen on two different files now after upgrading to version 2.2.1 on a Windows 10 x64 system.
Seems to happen when opening a .dxf file that has a hatched area that is in an active layer after having first made a minor edit to the file and saving it. 
I've attached a screenshot of the error, in the background is the hatched area. The error occurs after the hatched area is rendered but before the rest of the drawing is rendered. 

[image]

dxli, 2025-01-20:
Could you upload the dxf with this issue?

This is an issue during hatch processing.

veperr, 2025-01-20:
Sure, here's the dxf file this happened on. 

[Utility_wall_Detail.dxf.txt](https://github.com/user-attachments/files/18472226/Utility_wall_Detail.dxf.txt)

You'll need to rename the file extension to .dxf, github would't let me upload it as a .dxf file.

dxli, 2025-01-22:
This issue was caused by some border calculation optimization:  unless being forced, the entityContainer won't calculate the bounding boxes for invisible entities.

This causes some zero sized bounding boxes for hatch loop loop creation, in this bug scenario.

The fix: within the loop creation, force entityContainers to calculate bounding boxes.

u2fly, 2025-04-21:
On Linux latest `LibreCAD-2.2.2_alpha1-398-g65f3c5fa7.AppImage` failed:
- Ellipse (Axis) creation
```
call action Draw ellipse with axis
call action Draw ellipse with axis
librecad: /usr/include/boost/math/special_functions/ellint_rg.hpp:50: T boost::math::detail::ellint_rg_imp(T, T, T, const Policy&) [with T = long double; Policy = boost::math::policies::policy<boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy>]: Assertion `z >= y' failed.
Aborted
```

- MText font change
```
/usr/share/librecad/fonts
/usr/share/librecad/fonts
RS_Font::loadFont: Successfully opened font file: /usr/share/librecad/fonts/iso3098.lff
call action Draw Text
call action Draw Text
/usr/share/librecad/fonts
/usr/share/librecad/fonts
RS_Font::loadFont: Successfully opened font file: /usr/share/librecad/fonts/iso3098_i.lff
 : (       nan,        nan) vs (       nan,        nan)
delta: nan(ulp nan, nan(ulp: nan
librecad: /home/runner/work/LibreCAD/LibreCAD/librecad/src/lib/gui/render/rs_painter.cpp:1518: RS_Vector RS_Painter::toGui(const RS_Vector&) const: Assertion `!"toGui() failure"' failed.
Aborted

```

dxli, 2026-09-14:
Status on the current branches, checked with the attached `Utility_wall_Detail.dxf`:

- **The assertion:** fixed in v2.2.1.1 by 2a6697457. On v2.2.1 it took a hatch boundary with an edge drawn twice, on a frozen layer. The stud hatch in this file has its bottom side drawn out and back.
- **Two hatches dropped:** since v2.2.1.3, 2.2.1 opens this file without the stud and the solid scale bar, 2 of its 21 hatches. #2312's closure check rejects a boundary with an edge drawn twice. #2858 keeps both hatches and measures them right.
- **Wrong areas:** the area 2.2.1 reports is wrong for hatches with holes, such as hatch 110 here (4857.19 instead of 585). #2859 fixes that.
- **master:** opens the file with all 21 hatches and correct areas. Its `dxf2png` drew only the page and the border because of a colour conversion bug, which #2863 fixes.

Leaving this open until #2858 is merged.