LibreCAD/LibreCAD #2025: Plugin for CNC milling path.

Jullyfish, 2025-01-25:
There's not many tools for creating 2D cut with milling cutter compensation in Linux. I know only gcodetools plugin for Inkscape (I never used).

My suggestion is to make it work as follows:

1) Select polyline.
2) Set milling cutter diameter
3) Set compensation direction. Left or right.

[image]

sand1024, 2026-09-26:
@dxli 
Could it be related to your latest changes in offset? 
Or it's rather closer to the equidistant polyline command? It seems later command is more suitable, yet as far as I remember it does not support milling path (it is good to support both inside/outside modes) and so requires quite a deep rework.

Jullyfish, 2026-09-27:
I wrote this issue long time ago. I admit that my issue was partially incorrect because LibreCAD is CAD, not CAM. And I found [dxfplotter](https://github.com/panzergame/dxfplotter). dxfplotter can generate gcode with radius compensation.

So maybe better to close this issue as not planned.

Thank you for your time!

sand1024, 2026-09-27:
@Jullyfish 
well, that's quite popular feature actually, so probably it will be supported one day. 

And anyway - thank you for pointing to dxfplotter.

Jullyfish, 2026-09-27:
If forget about milling cutter radius, there's no "scale with reference" like in [AutoCAD](https://www.autodesk.com/support/technical/article/caas/sfdcarticles/sfdcarticles/How-to-resize-or-rescale-an-AutoCAD-drawing.html) as I know.

I think that first the `scale` feature should be improved. Then creating of milling path must be not part of LibreCAD. It may be some kind of macro or extension.