LibreCAD/LibreCAD #2029: Trim Two tool strange behavior

Questions2000, 2025-02-01:
BUG REPORT
This post is based on the following issue, that was corrected but seems to have generated a new issue.
https://github.com/LibreCAD/LibreCAD/issues/2002

### Expected Behavior
For "Trim Two" tool to highlight the section that is going to be removed.
I tested this out in an older version I had
LibreCAD-2.2.2_alpha1-234-gdb4ff4d9.AppImage
and this version seems to function as expected.

### Observed Behavior
In trying to help out by testing the issue mentioned in Issue Post 2002, I noticed that when that issue was corrected it now generates some strange behavior when using the "Trim Two" tool.
Based on the example shown in 2002 issue page, now if you use this tool, it does not appear to highlight the section of the line that is going to be removed, it ends up showing an angled yellow highlighted line, forming a triangle shape.

### Steps To Reproduce
If you follow the example mentioned on 2002 issues page
You basically have a "T" shape

-Activate "Trim Two" command
-Click on the LEFT side of the horizontal line
-Then when you bring cursor over to click on the vertical line
-You will notice that a yellow indicator angled line now appears starting from the left end point of the horizontal line to the bottom end point of the vertical line, creating a triangle shape

-In LibreCAD-2.2.2_alpha1-234-gdb4ff4d9.AppImage version I tested
the RIGHT section of horizontal line highlights yellow, indicating that that section of the line will be removed.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-256-g69766102.AppImage

sand1024, 2025-02-02:
please ignore this issue so far - previous fix was temporary, and one I've mentioned was not committed/merged yet.

dxli, 2025-02-04:
@sand1024 

For trim and trim2, the ancient logic is by the concept of a "trim point" is causing lots of trouble.

I suggest we redo all the logic:

For each entity, trim is done by the input:

1, limiting entity;
2, mouse pointer.

For trim2:

The behavior is consistent with trim, by using the other entity as the limiting entity.

Lots of work is needed, but this new framework should be much easier to prove correctness.

sand1024, 2025-03-29:
fixed in #2078

sand1024, 2025-04-23:
@Questions2000 
Please check on the latest 2.2.2 alpha and close if you could not reproduce it anymore