LibreCAD/LibreCAD #2030: dimension problem

jimmiehillman, 2025-02-10:
Having an annoying issue with Librecad 2.2.1 on Ubuntu 22.04 
When dimensions are inserted on a dimension layer (layer 1)only the last entered dimension can be selected.   Click on any other element or anywhere on the screen and the last entry is always selected.  Can't select any other element on the screen to modify or move  Only the last entry. 

I updated to the current version of Librecad and so blamed that for the issue.  But the software isn’t the problem.  The problem is the wetware.  Librecad works fine.

The novel way I managed to glitch the drawing.
I did a move sequence on some elements on layer 0, layer 0 active and layer 1 not visible.
That put the dimensions on layer 1 out of place so did a move on the elements that need to be moved and fumbled the process. In doing so managed to copy the elements onto layer 1 and layer 0  This seems to have linked layer 1 and 0 

Don’t know how I achieved that. Haven’t been able to do it again.  Tried a number of wrong ways but can’t break it yet. Don’t know what assorted functions and options were set at the time.  One of my better glitches Requires talent

Compared a good file with the bad file and it came up with $snapstyle.  sand1024 suggested it may involve snaps. Deleting the line and associated data didn’t fix it first try.  Look at it a bit more in time.

As an aside another thing I come up against from time to time. Create a block and delete it and later if you try to create a block with the same name an error message says it already exists.  Perhaps this is intended behavior

[dxf_file.zip](https://github.com/user-attachments/files/18732168/dxf_file.zip)

.

jimmiehillman, 2025-02-11:
Just discovered that if dimension arrow size is set to 0 in :- current drawing preferences - dimensions -: the problem happens.

dxli, 2025-02-12:
The root cause:

The distance to a solid area (the arrow heads) fails when the solid area is defined with degenerate points. When all contour points of a rs_solid are coincident, the test to find whether a point is within the contour always returns true. Therefore, the distance from a degenerate rs_solid to any point is zero, and this causing the first rs_dimension with zero arrow size being selected.

The fix ([1837fd3](https://github.com/LibreCAD/LibreCAD/commit/1837fd3547109791a4668f258d0e8a33ba5706b7):

fixed the rs_solid to point distance to support degenerate contours.

closed 2025-03-23 by dxli