LibreCAD/LibreCAD #2031: Implementation of an "AutoCAD" style of command issuing/drawing

samuelbuenofran, 2025-02-10:
if you have used AutoCAD before, when you begin typing a command, your cursor displays the text you are entering and upon hitting the enter key, you activate the corresponding command mode. You can also type in the length and dimensions of what you are drawing, for example say, a simple straight line.

Is this feature possible to implement in LibreCAD, or is it a patented/copyrighted thing from Autodesk?

sand1024, 2025-02-11:
well, if you I understand you correctly, you refer to the inputs/UI that are shown on top of drawing near the mouse cursor in AC? 

Or you've mentioned the mode of the command line input area? 

Could you please add a couple of screenshot to illustrate? 

Thank you.

sand1024, 2025-03-29:
@samuelbuenofran could you please expand on the expected functionality?

samuelbuenofran, 2025-04-08:
[image]

[image]

Although these two images were captured on the web version, they are also applicable to the desktop version. In essence, you may enter whatever command you like (for example, "ortho"), and the program will instantly react by carrying out the relevant operations linked to that command.
Additionally, AutoCAD allows you to enter the line's angle and distance, which is displayed next to the cursor whenever you attempt to create something.

sand1024, 2025-04-08:
@samuelbuenofran
Oh, thank you for additional info. 

>whatever command you like (for example, "ortho"), and the program will instantly react 

Hm.. but if you enter the command in Cmd widget - LibreCAD will also start react instantly. 
Or probably I've miss your point there? 

>which is displayed next to the cursor whenever you attempt to create something.

Well, as for parameters - they are currently shown not near to cursor, yet in Action' Options toolbar (usually it's located on top).   In LibreCAD it is also possible to enter line angle/distance...  While for "line" command there is small amount of parameters, some commands (like slice/divide or so) supports lot's of options, and putting them over drawing area will simply overload UI, I suppose.  

Potentially, one day in the future we may add similar way of entering parameters (directly on screen). However, this require quite a lot of development efforts, so I don't think it will be too soon. 

So far we have some approaching implementation in 2.2.2 branch, with better preview and information(dev alpha version is located there - https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.2_alpha-latest)

u2fly, 2025-04-10:
@sand1024

> Potentially, one day in the future we may add similar way of entering parameters (directly on screen). However, this require quite a lot of development efforts, so I don't think it will be too soon.

Also on "Entity Info" dockwidget it would be useful to be able edit changable values directly too.

sand1024, 2025-04-10:
@u2fly
Well, it's possible to edit properties via "Properties Edit" button in Entity Info widget toolbar. So far it just invokes properties edit dialog. 

Yes, this feature is useful too and is also somewhere in the todo list.  However, the editing of properties in such way is more powerful when it's applied to the set of entities, and that complicates implementation. Anyway, it's noted.

samuelbuenofran, 2025-04-14:
> [@samuelbuenofran](https://github.com/samuelbuenofran) Oh, thank you for additional info.
>
>
> Hm.. but if you enter the command in Cmd widget - LibreCAD will also start react instantly. Or probably [...]

I guess my way of explaining was somewhat really bad... The feature that is implemented in AutoCAD that I wish would be implemented in LibreCAD is the fact that you can click and type in the length, height, etc. of the line/arc and then press enter, which will make AutoCAD draw the line. This does not happen in LibreCAD.

sand1024, 2025-04-14:
@samuelbuenofran 
I'm still confused a bit. 

Considering line 2 points - you can try the following: 
1) Invoke the action (either via UI or via command line)
2) Enter start point of the line (again, either by providing the coordinate of the start point in the command widget, or just by left mouse click)
3) You'll be asked for the last of point of the line - you can simply enter second point as relative polar coordinates. 
Format is @len < angle (so @100<45 will define end point of line as point located at 45 degrees from the line's start point on distance which is 100). Basically - this is equivalent of entering line and angle.  

Action options toolbar for line 2 points does not include options indeed.  For other types of line-related actions (like Line Angle) 
there are such settings:

[image]

For Snake command, the set of options is even larger, and it's possible to enter length and angle there too: 

[image]

So I still confused... Or probably you mean some specific command?  Please let me know.

sand1024, 2026-09-26:
feature is implemented in relative point assistant - not as inplace edit, yet close enough.

closed 2026-09-26 by sand1024