LibreCAD/LibreCAD #2034: LibreCAD v2.2.1.1 crashes when opening dxf file

yazck, 2025-02-19:
Hi, 

[log-files.zip](https://github.com/user-attachments/files/18875380/log-files.zip)

Drawings were made on a dxf file produced from Autocad.
The dxf file size is 539 kB.
I am using Linuxmint v21.1 operating system. There is no problem with my use of libreoffice-v2.1.3 here.
I also test new versions from time to time.
When I saw the newly released version 2.2.1.1 a while ago, I did the tests again. The last 2 versions crashed in my environment.
I called it from the console in a clean Linuxmint v22.1 environment and did the following tests.

LibreCAD-2.2.0.2-x86_64.AppImage is Working
LibreCAD-v2.2.1-x86_64.AppImage is Broken
LibreCAD-v2.2.1.1-x86_64.AppImage is Broken

I cannot send the content of the dxf file. However, the relevant logs from the console are in the attached files.

yazck, 2025-02-20:
> did you try 2.2.2-alpha?

No I haven't tried it, if so how can I access it? If you send me the link I can try it.

dxli, 2025-02-20:
https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.2_alpha-latest

yazck, 2025-02-21:
> https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.2_alpha-latest

Ok, all the dxf files that were broken while opening the file opened properly. This version is successful 👍
The bug fixed, thank you for your support. 👍

dxli, 2025-02-24:
hi @yazck

The fix is now backported to the 2.2.1 branch. Please test with the latest releast candidate build.

yazck, 2025-02-24:
> hi [@yazck](https://github.com/yazck)
> 
> The fix is now backported to the 2.2.1 branch. Please test with the latest releast candidate build.
LibreCAD-v2.2.1.1-2-gb1531bbb-x86_64.AppImage file is working, Tested it's suitable, problem solved, thanks

closed 2026-09-26 by sand1024