LibreCAD/LibreCAD #2035: Arc and straight line closure problem

TiuMui, 2025-02-20:
When constructing an arc from the edge of a straight line, they do not close. This can be seen when increasing the intersection area.

[image]
[image]

dxli, 2025-02-21:
can you upload a minimum sample dxf with this issue?

TiuMui, 2025-02-21:
Sample file. When opened in version 2.2.0.2 there are no breaks, but in 2.2.1.1 there are breaks.

[sample.dxf.zip](https://github.com/user-attachments/files/18916939/sample.dxf.zip)

sand1024, 2025-02-22:
@TiuMui 
Yes, such behavior might happen for some specific zooms and arcs sizes. 

This is related to Qt-native method of arcs rendering, that may introduce un-exact drawing near to endpoints (it's just a rendering issue, as arc's coordinates, geometry/snapping are correct). 

In general, this may affect only rendering on screen. Actually, using of such method is a tradeoff between the speed of rendering and accuracy of rendering (actually, there should we quite specific conditions for noticing such behavior on real drawing). 

@dxli 
However,  in my later pull request (not merged yet) there is a setting that allows better control of the way for arcs rendering, and it's possible to specify whether arcs should be renderer using faster yet less exact method, or they should be rendered using line-based approximation (which eliminates such artefacts near to the arc's endpoints).

TiuMui, 2025-02-22:
A more informative example of the same file in two different versions: 2.2.0.2 and 2.2.1.1. Unfortunately, the gaps that are visible on the monitor are also visible after printing on paper.
Tested in Fedora and Linux Mint, a combination of the above versions of both Flatpak and Appimage from the GitHub repository.

[image]
[image]
[sample.dxf.zip](https://github.com/user-attachments/files/18925281/sample.dxf.zip)

dxli, 2025-02-23:
This is a serious known issue.

Should be a blocker for 2.2.1.2

sand1024, 2025-02-23:
@dxli Well, in my last pull request I've addressed this issue - by adding several ways for rendering arcs as well as options for that. 

in 2.2.2 version - that's also reproduceable with Qt-native drawing of arcs. 

I suppose for 2.2.1 - it's just possible to backport the line interpolation algorithm.  It is similar to ones included into 2.2.1, yet is slightly more efficient due to less amount of sin()/cos() calculations. 

Please take a look to RS_Painter::drawInterpolatedArc method (this one operates by GUI coordinates). 

https://github.com/LibreCAD/LibreCAD/blob/13f1077e61fa345568c56566e225e88c955491e3/librecad/src/lib/gui/render/rs_painter.cpp#L291

sand1024, 2025-04-23:
@TiuMui 
Please check whether in latest 2.2.2 alpha. 
It should be fixed now - it if is so, please close the ticket.

dxli, 2025-04-25:
Hi,

I would like to keep this one open:

The original issue (arc rendering) is fixed, but the similar issue exists for ellipses/elliptic arcs.

sand1024, 2025-04-25:
@dxli 
Oh, ok - yet probably it's better to have separate issue that is specific to elliptic arcs to avoid confusions?

sand1024, 2025-11-10:
@dxli 
It seems that the issues is still not fixed. Now after your fixes (commit 960acbbdee6a47314ea9f23f0d8a5947c68620f5 at 22/08/2025) from the situation seems to be even worse than before: if there is an allipse, and viewport is moved, the ellipse is just not rendered at all if part of the ellipse is out of the screen. 
 
Here is a small demo that illustrates how it looks like.. https://youtu.be/_bvavlogHuE

I suppose the version I've commited was more stable. 

Could you please provide me with more information about the issues you've mentione above `(The original issue (arc rendering) is fixed, but the similar issue exists for ellipses/elliptic arcs.` ? 

I can take a look on them base on my previous implementation, if it will be simpler..

sand1024, 2025-11-11:
@dxli 

and a couple of generic notes (not truly related to the previous comment):

`The performance issue is there because we need to render within 1 pixel even after zoomed in by 1 million. The issue was fixed by introducing visible rendering, i.e., splitting by intersection points.`

Just curious, did you take any measurements there? Of course, the clipping will be performed on the Qt level anyway, but what is the cost of it?

Might it be (at least theoretically) that clipping in Qt will be efficient enough and so we can neglect our own segments rendering/clipping and just rely on it?

At least for drawing lines - I suppose clipping in Qt should be quite efficient, and Qt will check for clipping anyway. 

So it might be that in some cases there will be **double checks** (our own + Qt-native).

I just haven't had a chance/time to dig deeper into Qt sources to check this; probably you have more info.

`precision of rendering`

Well, this is an issue to think about, I suppose - is it something that is absolutely important and mandatory? Of course, endpoints should be in the right positions.

However, if some entity is **drawn** with some tolerance and error at large zooms/sizes - is it really such an important issue?

Of course, given that the geometry and math for the entity is **correct** (so snapping/intersecting/transforming/exporting etc. works fine).

Actually, what I'm afraid of is that in order to support corner cases that occur not so often, the main algorithm may become more complicated and slower than it might be required.

So this is a question of finding a balance and a precision/speed trade-off, I think.

dxli, 2025-11-11:
@sand1024 

The performance issue I mentioned was due to rendering millions of segments outside of the viewport. It was quite bad, making the program not responding. However, that was the default behavior of rendering by line segments within 1 pixel without viewport clipping.

To get the visible portion of visible segments can get more complex.

I'm thinking about adding a warning first, before finding the solution:

1. find bounding box overlapping between the entity and viewport, if no overlap, skip rendering;
2. find intersections between the entity the viewport border, if there's intersection, but no visible segment found, generate an error. This appears to be the bug behavior here.