LibreCAD/LibreCAD #2036: Circle center point not being considered a valid move point when drag moving multiple selected items

Questions2000, 2025-02-20:
Was holding off making more posts until some of my other past posts were completed but they have gone beyond the main first page so unfortunately probably no interest in them.

### Expected Behavior
To have consistant behavior when performing drag move of multiple selected items when dragging from circle center point.

### Observed Behavior
If you select multiple items (one of them being a circle) and perform drag move near any of that circles 4 grip points all the selected items move.
However if you perform the same action using the center point of that circle, then only the circle moves and none of the other selected items move.

### Steps To Reproduce
-Draw 3 circles
-Select all 3 circles
-LMB click and drag near one of the circles 4 grip points to perform a mouse drag move
You will notice that all the circles move together as a group
-LMB click and drag from one of the circles center points to perform a mouse drag move
You will notice that now only that circle moves and the other circles remain where they are

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-263-g36f91f0e.AppImage

sand1024, 2025-02-22:
well, honestly speaking, I'm not sure that this is a bug... it might be that this is intentional behavior or at least interesting side effect. 

Indeed, moving by handle of the center point will move only that particular circle (and not the entire selection). 

But! Moving other handle of the circle (non-center one) will change the radius of that specific circle only (again, not radius of all circles within the selection). 

I'll investigate this further and check whether there are some historical reasons for such behavior.

Questions2000, 2025-02-28:
No I know, because the center point is an "imaginary" point it appears to be being ignored.
To me, if it is used to move just the circle itself, seems like it should also be used as a moving point for multi selected items.
As mentioned in post 2037, same thing occurs with ellipse center points, which was expected.
Also, correct in the context of drag moving I am referring to dragging near the actual grip points, which in DC uses the closest grip point as the dragging reference point, yes if you try to use the actual grip points the radius of that item will adjust.

Thank You

sand1024, 2025-03-29:
@Questions2000 
well, after some experiments, I decided that this is not a bug but quite a reasonable behavior. 

Actually, with moving by mouse if several entities selected there are two different policies: 

1) Moving starting from any entity within current selection using non-grip point - moves the entire selection. 

2) Moving grip point of entity within selection  - modifies just that specific entity WITHOUT affecting the selection. 

I've played more with this - and it seems that the current implementation is more convenient comparing to just moving the selection using any (grip/non-grip) point of entities. 

So I'd rather consider existing functionality as a "feature" than bug, so this issue may be closed. 

P.S - and this is functionality is supported with any entity, not only the circle. Just try to add a line and  include it into selection. 
Moving grip point of that line will affect the line only, while moving starting from non-grip point of line will move the entire selection.

closed 2025-05-10 by Questions2000