LibreCAD/LibreCAD #2037: Circle center point not being considered a valid copy point with drag copy action

Questions2000, 2025-02-20:
### Expected Behavior
When pefroming a drag copy action of a circle from its center point (when holding Ctrl) to perform a copy

### Observed Behavior
When performing a drag copy action of a circle from any of its 4 grip points (when holding Ctrl) the circle copies fine.
When performing a drag copy action of a circle from that circles center point (when holding Ctrl) the circle does not copy.
Center points are not considered to be valid points of copy.

### Similar Post
Here is another similar post regarding how circle center points are viewed
#2036 

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-263-g36f91f0e.AppImage

sand1024, 2025-02-22:
hm... indeed. ok, I'll take a look, thank you.

Questions2000, 2025-02-28:
Okay, thank you.
Not sure if helps but I recently tested and same thing occurs with "Ellipses" as well, which was expected.

Questions2000, 2026-10-10:
Tested in the following version and still occurs

### Operating System and LibreCAD Version
LibreCAD-2.2.2_alpha1-738-ga5f91b6b5-x86_64.AppImage
Compiled on: Oct 5 2026
System: Debian GNU/Linux 13 (trixie) (x11)