LibreCAD/LibreCAD #2038: Show guide lines when mouse cut or copy and then paste

Questions2000, 2025-02-20:
### Observed Behavior
-Select an item 
-Use Ctrl+X or Ctrl+C to cut or copy
-Click on a reference point
-Ctrl+V to paste
These two actions (cut / copy) I notice do not show a guide line indicator from that reference point as most tools do, such as a mouse drag move action does.
The "Move/Copy" tool which is a similar tool also shows guide line indicators for these same Move (Cut) and Copy actions

Should these Ctrl based actions get them as well?

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-263-g36f91f0e.AppImage

sand1024, 2025-02-22:
well, that' a good point. 

However, it might be that copy/paste is performed between different documents... in such case, guideline is not practical there. 
Yet even if it's copy/paste from the same document - there are some technical complexities for displaying original reference point.. It might be that a guide line will be added later, yet I'm not sure it will not be considered as a high priority thing.

Questions2000, 2025-02-28:
@sand1024
Okay, obviously up to you.

Mentioning because it seems that all other similar actions include them.
-Drag mouse move
-Move/Copy tool - moving and copying

When cutting or copying between different drawings, then obviously the reference point would be from other drawing so in these cases the guidelines would not be shown.

Thank You

sand1024, 2025-03-31:
@Questions2000 
well, I've played with support of such ref points on copy - and finally removed it. 

Aside of copying from other document - it looks quite misleading if the copy/paste operations are not subsequent. I.e - make copy, do something else, do paste.  

So yes, the copy/paste operation is a kind similar to the move/copy - yet it is different and unlike to later one, copy/paste allows to copy and paste be separated in time. 

Thus please close this issue.