LibreCAD/LibreCAD #2039: Allow reference point click to activate paste action when using Ctrl based move and copy actions

Questions2000, 2025-02-21:
### Observed Behavior
To perform Ctrl based actions the following steps are required
-Select items
-Hit Ctrl+X or Ctrl+C
-Click on reference point
-Hit Ctrl+V to paste (is this step needed)

### Expected Behavior
I was wondering, if the Ctrl+V step mentioned above is needed?
Maybe can make the clicking of the reference point to automatically apply the action pasting action instead.
Similar to how the "Move/Copy" tool does.

If not / no one agrees with this, okay to just close, was just an observation.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-263-g36f91f0e.AppImage

Questions2000, 2025-02-28:
@sand1024
Not sure if you are agreeing or not on this one?
If you are then can leave open, but if not agreeing guess can close this, up to you.

sand1024, 2025-03-30:
@Questions2000 
well, reviewed this scenario one more time - and yes, I still think that CTRL+V is needed, so please close this issue.

closed 2025-05-10 by Questions2000