LibreCAD/LibreCAD #2041: Move-Copy Tool - Allow Ctrl to perform copy actions

Questions2000, 2025-02-21:
### Expected Behavior
Move/Copy Tool
The `Ctrl` key does not appear to have any action linked to it.
Suggesting to allow holding of the `Ctrl` key when performing a "Move" action, to make copies of that item instead of moving them.
User can then hold down the `Ctrl` key and LMB click to make as many copies of that item that they want to make.

### Similar Post
I made the following post about adding a check box that would allow this copy action to occur without the use of `Ctrl` key
#2040
I think possibly both are valid in adding
Check box approach would be maybe if user required many copying actions and did not want to have to hold down `Ctrl` key for long periods of time.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-263-g36f91f0e.AppImage

sand1024, 2025-02-22:
yes, this is good point, I'll add support for CTRL there. Checkbox might be an overkill and lead to confusion in ui

sand1024, 2025-04-23:
@Questions2000 
Should be fixed in last update after PR https://github.com/LibreCAD/LibreCAD/pull/2114 (in 2.2.2 alpha branch - continuous build)

Please check and close if completed.

sand1024, 2026-09-26:
implemented already

closed 2026-09-26 by sand1024