LibreCAD/LibreCAD #2042: Add a Copy tool

Questions2000, 2025-02-21:
FEATURE REQUEST

I have always felt it would be nice to have separate tools for both `Move` and 'Copy'
The removing of the modify tool dialogs has really helped with this, but I feel that it could still be nice to offer a separate `Copy` tool.
Would allow user to establish a reference point and be able to create as many copies as they want, one at a time, by LMB clicking.
This way settings for the `Move` tool do not conflict with settings for the `Copy` tool.

`Move/Copy` tool would remain as is, I would still offer the `Copy` based parameters in that tool as it currently has.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-263-g36f91f0e.AppImage

sand1024, 2025-02-22:
well, that's quite a common request... I'll think about this yet still not sure I understand why it's better than current Move. 

So far it's possible to create a copy via ordinary drawing of selection (without the copy). 

Also, did you check the "Duplicate" command? Basically, it's a kind of Copy, yet it's rather intended for fast creation of duplicate for further modifications. 

Plus, regarding multiple copies, there are a couple of actions that might be more productive in some cases - it's "Paste Transform" and "Paste to Points"

Questions2000, 2025-02-28:
@sand1024

Ahh, I was hoping this would be an accepted request.
I think this one is really important so I am going to share a good amount of information about this, also because you asked.
Hopefully you will consider this after reviewing some of these points mentioned below.

-I will start off with one you mentioned, "it is quite a common request", so to me this means that there are multiple users mentioning it, even with all the current methods available, many users still see this as a desired tool to be added, and I agree.

-Move and Copy are actions that are used a great deal, very common tools

(Important One) 
-Being that Move and Copy actions are quite commonly used, making one "command" based tool to handle both, then requires a "toggle" in the form of a check box to determine what action the command is going to perform.  So when user activate the command, they always have to be aware of what "state" that command is in, is it move or is it copying?
If separate commands, user then will know exactly what command is being activated.

-Nice to have the current mouse/modifier based moving/copying approaches, but I personally view these as an added bonus that are available.
CAD programs being command line (widget) based, ALL tools use/have a "command" for them, so it is strange that two common tools require a completely different method of applying them.
(this was especially true when the Move/Copy tool was still dialog based, better now at least for Moving at the moment)
I feel the same way about the "Delete/Erase" command which for a while was unfortunately completely broken, so it required users to only have access to the "Delete" key.
(The "Erase" tool has been fixed although not all the way which I believe I made a post about in the past) (Will not get into why "Erase" command is important)

-Similar to command line based item mentioned above, user then has to remember different modifier/mouse based actions to achieve moving/copying

-Being able to just type in commands into the command line, especially if using keycodes can be quicker and more to the point.

-Separate copy tool as mentioned allows user to automatically make as many copies as they want, pasting that item from its reference point with each LMB click.
Approaches like "Paste Transform" I believe requires user to hit "Shift+Ctrl+V" combination the amount of times they want to have to copy that item.

-When making multiple copies of an item, if user has to make many of them, not always convenient to have to hold down modifier keys to have to copy items over and over.
I do not think there really are any ways to be able to just consistently make multiple copies over and over with each LMB click.
All require modifiers and most only can copy once and then have to reactivate that command each time.

-Move/Copy tool appears was originally designed using same point of view as something like the "mv" (move) command in Linux, where this command was also combined with the copying, but to me, in a CAD program the tasks are different actions.  Which is how many CAD/graphic design based programs offer these as separate tools, even though mouse/modifier actions would also allow for moving/copying.

-There are many tools that overlap use cases/actions with each other in LC, which is good, so why not the Move and Copy tools?

Okay, I will stop here :)

As you can see above, I still am hoping for the separate "Copy" command based tool to be added, but you mentioned
"Duplicate" command & "Paste To Points" I quickly tried these out and for me these did not do anything, I will have to test them out some more.
The "Paste Transform" as mentioned worked but requires hitting the key combination over and over for the amount of copies desired.  Side note, not sure I understand the name of this one, unless this command has other functions as well that I am not aware of yet (I can make a post about its name if you want me to)

Thank you to anyone who reads all of this