LibreCAD/LibreCAD #2043: Mouse drag copy allow holding down Ctrl to make as many copies as desired

Questions2000, 2025-02-21:
### Expected Behavior
When performing a mouse based drag copy by holding down `Ctrl` key it currently only allows for one copy.
Suggesting to make it so if user continues to hold down the `Ctrl` key that they can LMB click as many times as they want making a copy  of that selected item with each click.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-263-g36f91f0e.AppImage

sand1024, 2025-02-22:
yes, that makes sense, I'll add this.

sand1024, 2025-04-23:
@Questions2000 
Should be fixed in last update after PR https://github.com/LibreCAD/LibreCAD/pull/2114 (in 2.2.2 alpha branch - continuous build)

Please check and close if completed.

closed 2025-05-10 by Questions2000