LibreCAD/LibreCAD #2044: Allow Alt key to be used as a modifier for tools

Questions2000, 2025-02-21:
Okay, now that I have taken back a little of the main page real estate, I have one more item to mention.

Something I have been curious about is why LC only offers the two modifier keys `Ctrl` and `Shift`, but not `Alt`?
Would there be any interest in offering the `Alt` key to be used as a modifier for LC tool parameters, like `Ctrl` and `Shift` currently do?
This would allow tools to get one more extra action slot available for programmers to add to.
Unless there is some kind of conflict that the 'Alt' key would cause?

Thank You

sand1024, 2025-02-22:
well, there are several reasons
1) so far I don't see the need for an action/specific state where support of more than 2 modifiers is needed, if you have something in mind - please let me know. 
2) potentially, there might be issues with ALT on Mac

Questions2000, 2025-02-28:
@sand1024
First, thank you for responding to each of my posts.

Okay, was just curious.
The only one I can think of at the moment was for the Break/Divide, I had two modifier parameters I wanted to recommend but one of the modifiers was used by the item erase, that was one of the reasons why I agreed with the user who recommended the erase just work without the modifier.
Probably will not agree with the two uses of the modifier keys I have anyways, but maybe I will create a post for this in the future.

Did not know about the possible issue with Mac version.
Also including the use of the "Alt" modifier graphically will probably cause an issue in the Status toolbar.

Okay, for now I will close this post.

Thank You

closed 2025-02-28 by Questions2000