LibreCAD/LibreCAD #2045: Errors in font files

myhillman, 2025-02-22:
Using V2.2.1.1 I notice there are a few errors in the font files. They are mainly bulge values >1 or <-1, but there are a couple with malformed lines, e.g. Y value missing. I haven't checked them all, but standard.lff, unicode.lff, and cyrillic_ii.lff have problems. Someone needs to rigorously validate all the fonts.

myhillman, 2025-02-22:
My mistake. Bulge can be >1, but these are extreme cases. I have noticed a missing Y value in unicode.lff. The second vertex is "8,"

[ff0c] ，
8.,1.41;8.;7.19,-1.41

dxli, 2025-03-23:
@myhillman 

This font file has not been changed once after being created.

We don't have the source font file any more, and the original LFF developers are not reachable. I guess we have to fix them one by one.

The LFF implementation [rs_font.cpp](https://github.com/LibreCAD/LibreCAD/blob/master/librecad/src/lib/engine/document/fonts/rs_font.cpp) simply ignores incomplete coordinates (say, missing Y).

On the LFF font specification, here's the comments from Ries:

https://forum.librecad.org/Some-questions-about-the-LFF-fonts-td5715159.html

dxli, 2025-03-23:
Assuming the missing y-coordinates as 0 by commit 0bfd678

u2fly, 2025-04-16:
Latest LibreCAD 2.2.2.x removes some data on opening `unicode.lff` for editing (latest 2.2.1.x opens without such action), see CMD log:

```
Invalid objects removed: 24
Loaded document: C:\Program Files\LibreCAD\fonts\unicode.lff
```

u2fly, 2025-04-18:
@dxli @sand1024 

And more:

- On `../unicode.lff` opening it says `Invalid objects removed: 24` (related to https://github.com/librecad/librecad/issues/2108 ?).
- But if save opened file to separate LFF file and make a diff there is total messup with all Arc segments removed out. A sample of diff:
```Diff
 [0024] $
 2,0;2,9
-0,1.5;2.39,1,A.09;3.23,3.81,A.8;.77,5.19;1.61,8,A-.8;4,7.5,A-.09
+0,1.5;0.77,5.19
 
 [0025] %
 0,0;8,9
-3,7.5;0,7.5,A1;3,7.5,A1
-8,1.5;5,1.5,A1;8,1.5,A1
+3,7.5
+8,1.5
 
 [0026] &
-5,0;.74,6.69;3.23,6.65,A-1.86;.27,2.35;2.4,.3,A.85;6,3
+5,0;0.74,6.69;0.27,2.35;6,3
 
 [0027] '
-.5,7;1,9
+0.5,7;1,9
```

sand1024, 2025-04-18:
@u2fly 
The issue you've mentioned is not related to fonts, it's LibreCAD# 2108 indeed. Plus LibreCAD#2067

I'll commit PR with fix I suppose today or tomorrow the latest. 

Until that - it has no sense to save such file, it will be broken, I suppose.

u2fly, 2025-04-21:
@sand1024 
>I'll commit PR with fix I suppose today or tomorrow the latest.
>Until that - it has no sense to save such file, it will be broken, I suppose.

Opening LFF fonts is OK in latest built appimage artifact:

- https://github.com/dxli/LibreCAD/actions/runs/14570460950

But, saving to LFF still brokes fonts by removing arc elements/notes lines (started with `#`)/etc., even results in some characters totally removed (like  `[0028] (` and `[0029] )` trimmed to only single segment start point)

```diff
--- /home/user/Downloads/standard.lff
+++ /home/user/Downloads/standard2.lff
@@ -1,19 +1,18 @@
 # Format:            LibreCAD Font 1
 # Creator:           LibreCAD
+# Version:           2.2.2.3-alpha
 # Name:              ISO 3098-2
-# Version:           master
 # Encoding:          UTF-8
 # LetterSpacing:     3
 # WordSpacing:       6.75
 # LineSpacingFactor: 1
 # Created:           2011-09-25
-# Last modified:     2013-04-09
+# Last modified:     2025-04-21
 # Author:            Rallaz <rallazz@gmail.com>
 # License:           GPL v2 or later
-#
-# Notes:            - This font is based in the file
-#                     http://commons.wikimedia.org/wiki/File:ISO3098.svg
-#                     released as public domain.
+
+[fffd] �
+1,0;0,2;1,4;2,2;1,0
 
 [0021] !
 0,0;0,0.5
@@ -31,24 +30,24 @@
 
 [0024] $
 2,0;2,9
-0,1.5;2.39,1,A0.0935;3.235,3.81,A0.8;0.7685,5.19;1.6125,8,A-0.8;4,7.5,A-0.09
+0,1.5;0.7685,5.19
 
 [0025] %
 0,0;8,9
-3,7.5;0,7.5,A1;3,7.5,A1
-8,1.5;5,1.5,A1;8,1.5,A1
+3,7.5
+8,1.5
 
 [0026] &
-5,0;0.735,6.694;3.234,6.6472,A-1.8641;0.2660,2.3528;2.4,0.3,A0.8505;6,3
+5,0;0.735,6.694;0.266,2.3528;6,3
 
 [0027] '
 0.5,7;1,9
 
 [0028] (
-1,10;1,-1,A0.181818
+1,10
 
 [0029] )
-0,-1;0,10,A0.181818
+0,-1
 
 [002a] *
 2.8532,6;2.8532,3
@@ -72,36 +71,36 @@
 0,0;4,9
```