LibreCAD/LibreCAD #2047: Dimension line is separated from dimension arrows

Antonio-52, 2025-02-23:
Environment:
HW: HP HP Laptop 15-da0xxx
RAM: 8GB
LibreCAD:
Versione: v2.2.1.1
Compilatore: GNU GCC 9.4.0
Compilato in data: Feb 18 2025
Versione Qt: 5.15.2
Versione: 1.71.0
Sistema: Ubuntu 22.04.5 LTS

As soon as I draw a dimension, dimension line does not touch the arrow.
The distance from the dimension line end and the arrow is the same of the arrow size.

Antonio-52, 2025-02-24:
> Please upload a sample dxf file to show this issue.
>
> This appears to be related to particular dimension settings.

Hi!
I attached the DXF file.
The test drawing has the same settings of the previous version 2.2.0, that I worked with for a lot of drawings but I never observed such a problem.
--
Antonio Angelo

Inviato con l'email sicura [Proton Mail](https://proton.me/mail/home).

lordofbikes, 2025-02-24:
Attachment of emails does not work, you have to use GitHub to upload the sample.
Also DXF is not allowed to upload, you have to ZIP it or rename it, e.g. DRAWING.TXT.

Antonio-52, 2025-02-24:
[20250223_test_dxf.zip](https://github.com/user-attachments/files/18946127/20250223_test_dxf.zip)

dxli, 2025-02-25:
this is a regression due to a mistake in #2030 fixing.

Fixed by reverting the problematic commit:

Revert "Issue #2030: avoid zero size arrows": a5e47b505c259858453b0e0e74d7d90016992d79.

sand1024, 2026-09-26:
fixed

closed 2026-09-26 by sand1024