LibreCAD/LibreCAD #2049: Parallel building consumen too much RAM

Txikilinux, 2025-02-24:
Compiling this program with "$(MAKE) -j$(nproc)" takes more than 10gib RAM. Why? It seems that cmake builds in parallel with normal RAM consumption.

lordofbikes, 2025-02-24:
This is not an issue with LibreCAD code.

```$(MAKE) -j$(nproc)``` uses all available CPU cores and 10GB does not sound too much for me on a recent system. Of course, if you are on an older or limited system, this might be too much, but then you have to ask the compiler maintainer for appropriate options.

You can try to start with ```$(MAKE) -j2``` and rise the number to see the capabilities of your system.

Txikilinux, 2025-02-24:
I understand what you say, but libreCAD correctly builds with cmake with 8 cpu, working 100% all. And programs like Blender also build with 8 cores taking less than 4gib RAM in total. It's just LibreCAD that takes too much RAM (and makes the computer stall) with "make -j8".