LibreCAD/LibreCAD #2050: Change Move-Copy tools toolbar parameter labels

Questions2000, 2025-02-28:
Suggesting to change the Move-Copy tools toolbar parameter labels.
The labels currently being used are based on the labels that were used in the dialog box.
I personally think these are a little confusing.
Maybe consider changing
"Keep Originals" to "Make Copy" or "Create Copy"
and
"Multiple Copies" to something like "Array Copy" or "Create Array"
I knew what this one meant because it was in the Move/Copy dialog version, even knowing this, this still confused me because was expecting it to allow for multiple copies to be made one at a time with each LMB click.

Thank You

sand1024, 2025-02-28:
yes, I'm also not sure that is better naming there. 

Multiple Copies most probably should stay as it is was used earlier.  But adding "Create Copy" - will confuse even more together with "Multiple Copies"...   So current combination of labels is a trade-off, I suppose

Questions2000, 2025-03-09:
If I understand correctly it does not sound like interested in updating these labels, so I will close this issue.

Questions2000, 2026-09-19:
Going to create a new general post about this.

closed 2026-09-19 by Questions2000