LibreCAD/LibreCAD #2051: Modifier based cut & copy actions do not paste without zoom or pan action after

Questions2000, 2025-02-28:
### Observed Behavior
If you use Ctrl+X or Ctrl+C to cut or copy an item, after establishing a reference point, I noticed that you cannot just use Ctrl+V to paste those items.
You have to zoom or pan the drawing pane and then Ctrl+V to paste works.

Also if you try hitting Ctrl+V to paste again after this, it does not work, you have to zoom or pan again each time you want to paste another copy.

If possible, should just perform paste action without the need for zoom or panning.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-263-g36f91f0e.AppImage

sand1024, 2025-02-28:
hm... strange, yet I can't reproduce this. 

When you've mentioned you "can't use" - it means that you press CTRL+V and nothing happens?

But the shortcut just initiates paste operation, and mouse click defines insertion point and so completes the operation.  If there is no click - no insertion. 

So I'm not sure I understand this...

Questions2000, 2025-03-03:
@sand1024 
Thank you for reply, this one might be hard to find, but I was at least able to now get this to work.
Not sure if the information will be helpful but I can share what oddly fixed this.
Not sure how much I would look into this, up to you.
Might not be worth reading, but just wanted to document this information just in case this issue ever comes back again for me or any other users.

First I will just give a basic example of what the issue was

`Steps To Reproduce Issue:`
-Draw circle
-Select that circle
-Hit Ctrl+C to copy circle
-Click center point of circle as reference point
-Hit Ctrl+V to paste circle
When this paste hotkey is applied is where the issue could be experienced.
Expect to see that circle attached to mouse cursor by its center point in order to paste in new location.
However nothing would happen.

Would only work if doing any of the three items mentioned.
Because mouse was then still just a cursor, no tool activated...
-LMB click anywhere on screen to start a selection window, could hit Escape to cancel this action or just click again to complete the selection window on blank screen
-Could zoom in/out
-Could pan the drawing pane
Basically all of these are doing some kind of "refresh" that after they were performed would then allow the paste action to work but hitting Ctrl+V right after them

------------------------------------------------------------------------

`Steps I Performed That Oddly Fixed This`

-I first thought it might have been something with the version, so I tried the newest version available
`LibreCAD-2.2.2_alpha1-270-g8e25e848.AppImage`
which experienced same result

-Then I thought that I had been using the same .config/LibreCAD folder for a while so I just temporarily moved it elsewhere and started up LC which starts fresh settings.
At this point it worked, the issue was fixed.
So could just be that newer changes were made recently, similar in the past where must use fresh settings with newer versions of LC.

- In case the above is not true, I then thought I would try to help maybe narrow down if it was an actual setting that maybe was causing this.  I brought my OLD .config/LibreCAD folder back and the issue occurred again.

- I unchecked a couple settings and this still occurred, then I when I check the following setting, OFF, the issue was fixed.
`Show snap options in snap selection`
When I turned this setting back ON the issue occurred again.

-With that setting ON, I then thought to turn off ALL my snap icons at bottom of LC.
After doing this, the issue again was fixed.
I then tried turning ON each snap setting one at a time and no issues with any of them.
I unfortunately do not remember the exact layout of snaps that were ON when noticed the issue.
I tried a couple of the snaps layouts that I usually use but none generated this issue again.

At the moment I have not been able to regenerate the issue again, but if I do I will try to provide more information about my findings.

sand1024, 2025-03-31:
@Questions2000 

Thank you for such a detailed explanation.  
I've tried to reproduce it, but it seems "Show Snap options in Snap Selection Toolbar" does not affect such behavior, it seems that everything works as expected. 

But!

I suspect the situation was a bit different. Considering your example with Circle - 
1) if "Snap Center" mode is enabled
2) and you try to paste somewhere near of circle (of if the circle is the only entity) 
3) - it might be that snap is fired and since you try to copy the same circle, visually preview of copied circle may be not visible as is is placed exactly at the location of the original circle. 

This is valid situation, yet actually not too convenient - as during the Copy operation, "snap to center" is convenient for selecting center point of the circle. 

But on Paste operation - "snap to center" may be misleading...  Potentially, it is possible to add support of some modifier into Paste action that will allow, for example "free snap" during the insertion point selection phase.  

Not sure, however, whether it will be more convenient than just disabling "snap to center"...