LibreCAD/LibreCAD #2052: Why no DXF thumbnails or file preview in file open?

jpagliaccio, 2025-02-28:
It would be great to see thumbnails for the DXF files when opening files from the app. I realize this happens at the OS level but it would be very handy to include the functionality with LibreCAD. There is a Python library called ezdxf that generates thumbnails for DXF circles and lines that works. Has anybody tried to implement this?

The other option would be to add the functionality to the LibraCAD file open dialog as a DXF file preview the same way the Inkscape does it. I use the "native OS file open dialog" setting on Linux, but enhancing the LibraCAD file open dialog would e a good option too.

sand1024, 2025-03-05:
Well, we may support generation of thumbnails on save of dxf from LibreCAD. 
We can also may add support of thumbnails that are part of DXF, so if file is created by other program - it will be visible on open. 

However, we hardly can support generation of thumbnails on the fly when file is opening if there is no thumbnail exists in dxf. 
The reason is quite simple - to generate thumbnail in such case it's necessary to open the file and render it, this may be quite a long operations. 

For template drawings, that are supported by Library widget - thumbnails are generated.

jpagliaccio, 2025-03-06:
I hacked up a test in Python using the ezdxf library and Matplotlib before I saw your reply. Works for lines, circles, arcs, and polylines. A little slow but it does help when looking for files. And yes, creating the thumbnails on save would be most efficient, especially if you know geometry has changed rather than just the view has changed.      

Ah. You already have the code from the template thumbnails. That's great. Let me know if I can help or test.   

[image]
Not a thumbnail. Just an image of the DXF from the test program stored with same filename + .dxf.

sand1024, 2025-03-06:
hm... well, LibreCAD actually DO read and render dxf files by itself - why it should rely on exdxf? 

Adding thumbnails on save is not an issue, yet anyway it will be just partial solution - it will not work for dxf files that were not previously opened and saved by LibreCAD.  
Showing thumbnails for DXF that includes 90 and 310 codes (THUMBNAILIMAGE) - well, it's also possible. 

But adding thumbnails on open/browsing directory - it's hardly will be implemented, sorry - it will be too slow. 
 
Also, as temporary solution, for Window it's possible to rely on external utilities, like this one:
https://apps.microsoft.com/detail/9nqddnr3drn0

Probably similar tool exists for other platform too.