LibreCAD/LibreCAD #2054: No notarization for LibreCAD in Mac OS?

Tagbert, 2025-03-02:
I tried to install LibreCAD-v2.2.1.1.dmg and when I tried to run it, I got this message. From what I can tell, that is probably because this package has not gone through Apple notarization. Is that normal for a LibreCAD build? I don't want to have to disable all security to be able to run this app.

[image]

dxli, 2025-03-02:
https://support.apple.com/en-ca/102445

Tagbert, 2025-03-03:
Thanks for the link. I guess that answers my question that LibreCAD is not notarized. I had wanted to verify that before I use a method to circumvent security and run it anyway. It might be nice if there was a note in the installation notes mentioning this.