LibreCAD/LibreCAD #2058: Problem: View -> Statusbar (Ctrl+I)

BenMartinBox, 2025-03-04:
LibreCAD
Version: 2.2.1.1
Compiler: GNU GCC 14.2.0
Compiled on: Feb 19 2025
Qt Version: 5.15.15
Boost Version: 1.80.0
OS: Linux Mint 22.1 Cinnamon - version 6.4.8 - Kernel 6.8.0-54-generic - Display server X11.

Problem: View -> Statusbar (Ctrl+I).
I disable the Statusbar and save my "draw template" and the "Application preferences" as I like it.
On next restart, the "Statusbar" is back "ON" again.

It seems to be "ON" by default and the user can't deactivate it by default since the state is not saved as a preference.
Any chance to include this option on the save "Application preferences" (Disable Statusbar on start)?

BenMartinBox, 2025-03-05:
Much appreciated. Thank you very much.

sand1024, 2025-04-23:
@BenMartinBox 
Should be fixed in last update after PR https://github.com/LibreCAD/LibreCAD/pull/2114 (in 2.2.2 alpha branch - continuous build)

Please check and close if completed.

sand1024, 2026-09-26:
state of status bar is persistent

closed 2026-09-26 by sand1024