LibreCAD/LibreCAD #2059: Problem: Tool bar Button/Icons remain "selected" in cascade.

BenMartinBox, 2025-03-04:
LibreCAD
Version: 2.2.1.1
Compiler: GNU GCC 14.2.0
Compiled on: Feb 19 2025
Qt Version: 5.15.15
Boost Version: 1.80.0
OS: Linux Mint 22.1 Cinnamon - version 6.4.8 - Kernel 6.8.0-54-generic - Display server X11.

Problem: Tool bar Button/Icons remain "selected" in cascade.

Select the line tool "2 points" and then select down the list.
Notice that the previous icon/button remains selected, adding to the selection stack.
(See attached image for better understand the issue).

[image]

sand1024, 2025-03-05:
@dxli I can't reproduce this - yet I'm on 2.2.2 sources.. can you please check?  Might it be due to last changes in event handler ( 399878877058cc75415c10256c06777f1f5a2e86)? 

I remember that I've noticed and fixed similar behavior several months ago. It was reproduceable if dock widgets for left panel are visible (so action's buttons are always visible). Probably something similar is here..

BenMartinBox, 2025-03-05:
Hi, yes the problem continues being reproducible in version 2.2.1. from my side, both in "Linux Mint" and "Windows 10" (same PC - dual boot).
The GPU is a simple old Nvidia Quadro K620 - 2Gb RAM (most recent drivers). 

To test, I downloaded "LibreCAD" Version: 2.2.2_alpha1-270-g8e25e848
Compiler: GNU GCC 11.2.0
Compiled on: Mar  1 2025
Qt Version: 6.6.3
Boost Version: 1.87.0
System: Windows 10 Version 22H2

In this Version: 2.2.2 the problem is solved, so I guess there is no need to waste time and resources trying to fix this specific issue on version 2.2.1. Besides, it is not "problematic", meaning that it does not "break" or stop the user from using the app.

Yet, in such tests with the 2.2.2 Alpha, I noticed another situation, that I'm not sure if it is already identified:

If "Windows 10"  Color scheme is set to "Dark" and if in "LibreCAD"  the user changes the Style (on Widget Options) to "Windows" or "Fusion" instead of "windowsvista", the "LibreCAD" buttons /icons) become almost unreadable (see attached image for clarification).

NOTE: Didn't test this in Linux Mint (Dark color scheme).

[image]

sand1024, 2025-04-23:
@BenMartinBox 
Should be fixed in last update after PR https://github.com/LibreCAD/LibreCAD/pull/2114 (in 2.2.2 alpha branch - continuous build)

Please check and close if completed.

sand1024, 2026-09-26:
duplicate, solved

closed 2026-09-26 by sand1024