LibreCAD/LibreCAD #2060: Files are marked dirty (modified) when opening them, before any modifications are made

jpagliaccio, 2025-03-06:
Files are marked dirty (modified) when opening them, before any modifications are made.  

It would be more natural to mark them as modified only after a geometry change. Likewise with a view change. It should be up to the user to save after a view change. Maybe a new preference. 

The issue for the user, it leads to (over) saving files just to be sure to save a change if there was one. It , and messes up your file timestamps. Especially true if you have multiple files open, and annoying if you just want to open, view, and close a file.      

LibreCAD
Version: 2.2.0.2
Compiler: GNU GCC 14.0.1
Compiled on: Jan 25 2024
Qt Version: 5.15.15
Boost Version: 1.83.0
System: Fedora Linux 40 (Workstation Edition)

lordofbikes, 2025-03-06:
This is probably due to auto zoom on open, an application preference.
Can you check in _Application Preferences -> Defaults_ if _Automatically zoom drawing when opening_ is checked.

The view is stored in the DXF and thus zooming on open modifies the drawing.
This is discussed regularly and intended behavior.

sand1024, 2025-03-06:
yes, most probably this is due to zoom. 
In 2.2.2 branch there will be a setting that will disable making file as "modified" if only zoom/pan operations were performed (i.e. no actual editing).

sand1024, 2025-03-18:
addressed by #2032

sand1024, 2025-04-23:
@jpagliaccio 
Should be fixed in last update after PR https://github.com/LibreCAD/LibreCAD/pull/2114 (in 2.2.2 alpha branch - continuous build)

An option for this behavior is added to Application Preferences. 

Please check and close if completed.

sand1024, 2026-09-26:
setting was added that allows to move in and zoom without affecting modification status

closed 2026-09-26 by sand1024